mpromonet/v4l2rtspserver #266: Compile Error on Raspberry Pi 4

Qamada, 2022-01-28:
What should I do to avoid the compile error on the below.

**Compile error**
`$ make
[  4%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/V4l2RTSPServer.cpp.o
In file included from /home/pi/git/mpromonet/v4l2rtspserver/src/V4l2RTSPServer.cpp:20:
/home/pi/git/mpromonet/v4l2rtspserver/inc/V4l2RTSPServer.h: In constructor ‘V4l2RTSPServer::V4l2RTSPServer(short unsigned int, short unsigned int, int, unsigned int, const std::__cxx11::list<std::__cxx11::basic_string<char> >&, const char*, const string&, const char*)’:
/home/pi/git/mpromonet/v4l2rtspserver/inc/V4l2RTSPServer.h:41:35: error: ‘class RTSPServer’ has no member named ‘setTLSState’
   41 |                     m_rtspServer->setTLSState(sslkeycert, sslkeycert, true, true);
      |                                   ^~~~~~~~~~~
make[2]: *** [CMakeFiles/libv4l2rtspserver.dir/build.make:212: CMakeFiles/libv4l2rtspserver.dir/src/V4l2RTSPServer.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:117: CMakeFiles/libv4l2rtspserver.dir/all] Error 2
make: *** [Makefile:182: all] Error 2`

**To Reproduce**
Steps to reproduce the behavior:
1. sudo apt-get update
2. sudo apt-get upgrade
3. sudo apt-get install liblivemedia-dev
4. git clone https://github.com/mpromonet/v4l2rtspserver.git
5. cd v4l2rtspserver
6. cmake .
7. make

**Expected behavior**
No build error.
Last year, I have done successful compile for v4l2rtspserver on the same raspberry pi 4.

**Environment**
Raspberry Pi 4
$ u-name -a
  Linux raspberrypi 5.10.92-v7l+ #1514 SMP Mon Jan 17 17:38:03 GMT 2022 armv7l GNU/Linux
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye

**What I've tried as workaround**
I read issue #133 and #134 , and then tried `cmake LIVE555URL=https://download.videolan.org/pub/contrib/live555/live.2019.03.06.tar.gz .`
, but the same compiler error showed again.

mpromonet, 2022-01-28:
Hi,

This is a problem of compatibility with old live555. Setting the LIVE555URL doesnot change, because liblivemedia-dev is installed. I will fix this.

Best Regards,
Michel.

closed 2022-01-28 by mpromonet

Qamada, 2022-01-28:
> Hi,
> 
> This is a problem of compatibility with old live555. Setting the LIVE555URL doesnot change, because liblivemedia-dev is installed. I will fix this.
> 
> Best Regards, Michel.

Hi Michel,

I solved this issue as the below.

1.  sudo apt-get remove liblivemedia-dev
2. Install Live555 as http://www.live555.com/liveMedia/#config-unix says, but link error occurred, and even after I edit CMakeFiles/v4l2rtspserver.dir/link.txt , I decreased the number of link errors, but I wasn't able to compile without any errors.
3. I removed Live555 at all.
4. `cmake .`  again for v4l2rtspserver because I read one of workaround on #244 .
5. `make` and completed building without any errors.

Thanks,

mpromonet, 2022-01-28:
Hi,

Your first point should be enouph, the cmake is downloading latest live555 release (after a cmake cache cleaning for instance using `git clean -x -f -d`)

Best Regards,
Michek?

commit bf53d0e1 (2022-01-28): fix #266: use setTLSState depending on live555 version