mpromonet/v4l2rtspserver #267: Image transport defaults to TCP

pepeRossRobotics, 2022-03-07:
**Describe the bug**
The default data transport if TCP

**To Reproduce**
Steps to reproduce the behavior:
1. Run `v4l2rtspserver -W 640 -H 480 -F 30 -Q 2 -P 8554 /dev/video0`
2. In another computer in the same network open the stream i.e. `vlc rtsp://192.168.1.158:8554/unicast`
3. in that computer run `netstat` 
**Expected behavior**
Is expected that the result of NetStat should be something like this 
`udp        0      0 PepeX1C7.lan:58444      CM4.lan:8554            ESTABLISHED`

but it is this:
`tcp        0      0 PepeX1C7.lan:58444      CM4.lan:8554            ESTABLISHED`

**Screenshots**

[image]
[image]
```
root@CM4:/home/pi/v4l2rtspserver# v4l2rtspserver -W 640 -H 480 -F 30 -Q 2 -P 8554 /dev/video0
log level:500

[NOTICE] /home/pi/v4l2rtspserver/main.cpp:294
	Version: 0.2.4-8-g79a13bd live555 version:2022.01.06
[NOTICE] /home/pi/v4l2rtspserver/src/V4l2RTSPServer.cpp:37
	Create V4L2 Source.../dev/video0
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:133
	driver:bm2835 mmal capabilities:85200005 mandatory:4000001
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:136
	/dev/video0 support capture
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:138
	/dev/video0 support read/write
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:139
	/dev/video0 support streaming
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:225
	/dev/video0:H264 size:640x480 bufferSize:307200
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:246
	fps:1/30
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:247
	nbBuffer:1
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:49
	Device /dev/video0
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:73
	Device /dev/video0 nb buffer:10
[NOTICE] /home/pi/v4l2rtspserver/src/V4l2RTSPServer.cpp:62
	Create Source .../dev/video0
[NOTICE] /home/pi/v4l2rtspserver/inc/BaseServerMediaSubsession.h:49
	format:video/H264
[NOTICE] /home/pi/v4l2rtspserver/src/V4L2DeviceSource.cpp:96
	begin thread
[NOTICE] /home/pi/v4l2rtspserver/inc/V4l2RTSPServer.h:80
handleCmd_SETUP:SETUP rtsp://192.168.1.158:8554/unicast/track1 RTSP/1.0
Transport: RTP/AVP/TCP;unicast;interleaved=0-1
CSeq: 3
User-Agent: Lavf58.29.100

```

**Desktop (please complete the following information):**
 - OS Ubuntu running via docker container
 - The host computer is a RaspberryPi CM4

mpromonet, 2022-03-07:
Hi,

You may look to the RTSP configuration of VLC, that is probably configure to ask for RTP over TCP.

Best Regards,
Michel.

pepeRossRobotics, 2022-03-08:
I checked that VLC is configured to start with UDP, and still the transport using `netstat` tells me the system is usinc TCP.

Is there any flag that I could use to force the v4l2rtsp to be sent UDP?

[image]

mpromonet, 2022-03-08:
Hi,

This is the opposite, v4l2rtspserver is a server and answer to the client that asked for an RTP over TCP stream, as you can see in your log in the RTSP SETUP message. You can try with openRTSP.

Best Regards,
Michel.

pepeRossRobotics, 2022-03-15:
Hello,
I have tried without sucess to set up the VLC player to ask for the UDP transport.
I just tried with openRTSP and seems to be the sae:
```
handleCmd_SETUP:SETUP rtsp://192.168.10.254:8554/unicast/track1 RTSP/1.0
CSeq: 4
User-Agent: LibVLC/3.0.9.2 (LIVE555 Streaming Media v2020.01.19)
Transport: RTP/AVP;unicast;client_port=33678-33679

handleCmd_SETUP:SETUP rtsp://192.168.10.254:8554/unicast/track1 RTSP/1.0
CSeq: 4
User-Agent: openRTSP (LIVE555 Streaming Media v2020.01.19)
Transport: RTP/AVP;unicast;client_port=33468-33469
```

the first one is using VLC and the secondone RTSP, bot are using TCP. Could this be something to do with the camera that I am using?

mpromonet, 2022-03-15:
Hi,

RTP/AVP;unicast;client_port=33468-33469 is UDP (port are RTP and RTCP port). RTP over TCP have a transport like RTP/AVP/TCP (like in your initial post).

Best Regards,
Michel.

pepeRossRobotics, 2022-03-15:
Many thanks, that makes sense, sorry for all the time spend in such a silly thing. My bad.

closed 2022-03-15 by pepeRossRobotics

mpromonet, 2022-03-15:
You are welcome.