mpromonet/v4l2rtspserver #268: Error

bibitocarlos, 2022-03-19:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
When i start v4l2rtspserver pn my pi, latest upgrade, i get :
`v4l2rtspserver /dev/video0 -r -W 1920 -H 1080 -F 30
log level:500

[NOTICE] /home/pi/v4l2rtspserver/main.cpp:297
        Version: 0.3.0-2-g34dd957 live555 version:2022.02.07
[NOTICE] /home/pi/v4l2rtspserver/src/V4l2RTSPServer.cpp:37
        Create V4L2 Source.../dev/video0
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:133
        driver:bm2835 mmal capabilities:85200005 mandatory:1000001
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:136
        /dev/video0 support capture
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:138
        /dev/video0 support read/write
[NOTICE] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:139
        /dev/video0 support streaming
[ERROR] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:207
        /dev/video0: Cannot set format:H264 Invalid argument
[ERROR] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:207
        /dev/video0: Cannot set format:MJPG Invalid argument
[ERROR] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:207
        /dev/video0: Cannot set format:JPEG Invalid argument
[ERROR] /home/pi/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:83
        Cannot init device:/dev/video0`

`v4l2-ctl --list-devices
bcm2835-codec-decode (platform:bcm2835-codec):
        /dev/video10
        /dev/video11
        /dev/video12
        /dev/video18
        /dev/media1

bcm2835-isp (platform:bcm2835-isp):
        /dev/video13
        /dev/video14
        /dev/video15
        /dev/video16
        /dev/media0

mmal service -2143965256.0 (platform:bcm2835-v4l2-0):
        /dev/video0
`

` v4l2-ctl --list-formats
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'YU12' (Planar YUV 4:2:0)
        [1]: 'YUYV' (YUYV 4:2:2)
        [2]: 'RGB3' (24-bit RGB 8-8-8)
        [3]: 'JPEG' (JFIF JPEG, compressed)
        [4]: 'H264' (H.264, compressed)
        [5]: 'MJPG' (Motion-JPEG, compressed)
        [6]: 'YVYU' (YVYU 4:2:2)
        [7]: 'VYUY' (VYUY 4:2:2)
        [8]: 'UYVY' (UYVY 4:2:2)
        [9]: 'NV12' (Y/CbCr 4:2:0)
        [10]: 'BGR3' (24-bit BGR 8-8-8)
        [11]: 'YV12' (Planar YVU 4:2:0)
        [12]: 'NV21' (Y/CrCb 4:2:0)
        [13]: 'RX24' (32-bit XBGR 8-8-8-8)`

**Expected behavior**
To work ^^

**Desktop (please complete the following information):**
 - Raspbian
 - Firefox

mpromonet, 2022-03-19:
Hi,

It seems some arguments are not correct, mayby framerate or geometry ? 
Increasing the loglevel may help you.
You can also setting the capture format before running v4l2rtspserver and start it with `-f` without parameter, then it will not try to set format and use the current one.

Best Regards,
Michel.

bibitocarlos, 2022-03-19:
Dmesg 
`[76933.634403] bcm2835_mmal_vchiq: timed out waiting for sync completion
[76933.634455] bcm2835-v4l2-0: vidioc_s_fmt_vid_cap: failed to setup mmal components: -62
[76968.194375] bcm2835_mmal_vchiq: timed out waiting for sync completion
[76971.234416] bcm2835_mmal_vchiq: timed out waiting for sync completion
[76974.274541] bcm2835_mmal_vchiq: timed out waiting for sync completion
[76977.314411] bcm2835_mmal_vchiq: timed out waiting for sync completion
[77026.034432] bcm2835_mmal_vchiq: timed out waiting for sync completion
[77029.074433] bcm2835_mmal_vchiq: timed out waiting for sync completion
[77032.114365] bcm2835_mmal_vchiq: timed out waiting for sync completion
[77035.154410] bcm2835_mmal_vchiq: timed out waiting for sync completion
[78114.190630] bcm2835_mmal_vchiq: timed out waiting for sync completion
[78114.190678] bcm2835_mmal_vchiq: vchiq_mmal_component_init: failed to create component -62 (Not enough GPU mem?)
[78114.190700] bcm2835-codec bcm2835-codec: bcm2835_codec_create_component: failed to create component ril.video_encode
[81128.741911] bcm2835_mmal_vchiq: timed out waiting for sync completion
[81128.741964] bcm2835_mmal_vchiq: vchiq_mmal_component_init: failed to create component -62 (Not enough GPU mem?)
[81128.741982] bcm2835-codec bcm2835-codec: bcm2835_codec_create_component: failed to create component ril.video_encode
[81135.221928] bcm2835_mmal_vchiq: timed out waiting for sync completion
[81159.301777] bcm2835_mmal_vchiq: timed out waiting for sync completion
[81159.301824] bcm2835_mmal_vchiq: vchiq_mmal_component_init: failed to create component -62 (Not enough GPU mem?)
[81159.301840] bcm2835-codec bcm2835-codec: bcm2835_codec_create_component: failed to create component ril.video_encode
[81174.101830] bcm2835_mmal_vchiq: timed out waiting for sync completion
[81174.101879] bcm2835_mmal_vchiq: vchiq_mmal_component_init: failed to create component -62 (Not enough GPU mem?)
[81174.101902] bcm2835-codec bcm2835-codec: bcm2835_codec_create_component: failed to create component ril.video_encode
[81308.981620] bcm2835_mmal_vchiq: timed out waiting for sync completion
[81308.981673] bcm2835_mmal_vchiq: vchiq_mmal_component_init: failed to create component -62 (Not enough GPU mem?)
[81308.981696] bcm2835-codec bcm2835-codec: bcm2835_codec_create_component: failed to create component ril.video_encode
[81332.021625] bcm2835_mmal_vchiq: timed out waiting for sync completion
[81332.021676] bcm2835_mmal_vchiq: vchiq_mmal_component_init: failed to create component -62 (Not enough GPU mem?)
[81332.021698] bcm2835-codec bcm2835-codec: bcm2835_codec_create_component: failed to create component ril.video_encode
`

mpromonet, 2022-03-19:
"Not enough GPU" may help you ?

bibitocarlos, 2022-03-19:
Updated everything.
Add to boot 
"gpu_mem=256
dtoverlay=vc4-fkms-v3d"

New error 

` 70.515176] bcm2835-v4l2-0: Failed to enable encode tunnel - error -3
[   70.515203] ------------[ cut here ]------------
[   70.515276] WARNING: CPU: 2 PID: 1041 at drivers/media/common/videobuf2/videobuf2-core.c:1548 vb2_start_streaming+0x108/0x178 [videobuf2_common]
[   70.515287] Modules linked in: sha256_generic cfg80211 rfkill 8021q garp stp llc vc4 cec drm_kms_helper snd_soc_core snd_compress snd_pcm_dmaengine syscopyarea sysfillrect sysimgblt fb_sys_fops raspberrypi_hwmon snd_bcm2835(C) snd_pcm bcm2835_codec(C) v4l2_mem2mem bcm2835_isp(C) snd_timer bcm2835_v4l2(C) snd bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common vc_sm_cma(C) videodev mc fixed uio_pdrv_genirq uio i2c_dev drm drm_panel_orientation_quirks backlight fuse ip_tables x_tables ipv6
[   70.515708] CPU: 2 PID: 1041 Comm: v4l2rtspserver Tainted: G         C        5.10.103-v7+ #1530
[   70.515714] Hardware name: BCM2835
[   70.515721] Backtrace: 
[   70.515748] [<809f51d8>] (dump_backtrace) from [<809f5568>] (show_stack+0x20/0x24)
[   70.515761]  r7:ffffffff r6:00000000 r5:60000013 r4:80fe5ef0
[   70.515777] [<809f5548>] (show_stack) from [<809f977c>] (dump_stack+0xcc/0xf8)
[   70.515794] [<809f96b0>] (dump_stack) from [<8011ee94>] (__warn+0xfc/0x114)
[   70.515808]  r10:7f36e218 r9:00000009 r8:7f2821ec r7:0000060c r6:00000009 r5:7f2821ec
[   70.515816]  r4:7f286f90 r3:80f05050
[   70.515830] [<8011ed98>] (__warn) from [<809f5bb8>] (warn_slowpath_fmt+0x70/0xd8)
[   70.515841]  r7:0000060c r6:7f286f90 r5:80f05008 r4:00000000
[   70.515877] [<809f5b4c>] (warn_slowpath_fmt) from [<7f2821ec>] (vb2_start_streaming+0x108/0x178 [videobuf2_common])
[   70.515892]  r9:7f1800e8 r8:81433388 r7:00000001 r6:81433538 r5:81433638 r4:ffffffff
[   70.515945] [<7f2820e4>] (vb2_start_streaming [videobuf2_common]) from [<7f2828c8>] (vb2_core_streamon+0x90/0x180 [videobuf2_common])
[   70.515956]  r7:00000001 r6:80f05008 r5:94834000 r4:81433538
[   70.516016] [<7f282838>] (vb2_core_streamon [videobuf2_common]) from [<7f2c0030>] (vb2_streamon+0x24/0x64 [videobuf2_v4l2])
[   70.516025]  r5:94834000 r4:00000001
[   70.516062] [<7f2c000c>] (vb2_streamon [videobuf2_v4l2]) from [<7f2c00bc>] (vb2_ioctl_streamon+0x4c/0x50 [videobuf2_v4l2])
[   70.516288] [<7f2c0070>] (vb2_ioctl_streamon [videobuf2_v4l2]) from [<7f180114>] (v4l_streamon+0x2c/0x30 [videodev])
[   70.516299]  r5:40045612 r4:7f2c0070
[   70.516489] [<7f1800e8>] (v4l_streamon [videodev]) from [<7f1851c0>] (__video_do_ioctl+0x234/0x460 [videodev])
[   70.516499]  r5:40045612 r4:81433058
[   70.516685] [<7f184f8c>] (__video_do_ioctl [videodev]) from [<7f185be4>] (video_usercopy+0x138/0x5f0 [videodev])
[   70.516699]  r10:00000000 r9:93fede3c r8:00000000 r7:80f05008 r6:00000000 r5:40045612
[   70.516706]  r4:40045612
[   70.516892] [<7f185aac>] (video_usercopy [videodev]) from [<7f1860bc>] (video_ioctl2+0x20/0x24 [videodev])
[   70.516906]  r10:00000005 r9:93fec000 r8:00000000 r7:94834000 r6:94834000 r5:80f05008
[   70.516913]  r4:7f18609c
[   70.517098] [<7f18609c>] (video_ioctl2 [videodev]) from [<7f17e1f4>] (v4l2_ioctl+0x4c/0x60 [videodev])
[   70.517202] [<7f17e1a8>] (v4l2_ioctl [videodev]) from [<8034c0d0>] (sys_ioctl+0x1d4/0x8ec)
[   70.517212]  r5:80f05008 r4:40045612
[   70.517227] [<8034befc>] (sys_ioctl) from [<80100040>] (ret_fast_syscall+0x0/0x28)
[   70.517236] Exception stack(0x93fedfa8 to 0x93fedff0)
[   70.517248] dfa0:                   045b7000 00000000 00000005 40045612 7e9b2150 00000005
[   70.517260] dfc0: 045b7000 00000000 00000009 00000036 00000000 00000000 76fa4000 7e9b2184
[   70.517270] dfe0: 00117138 7e9b20ec 000ecd40 76ce21ac
[   70.517283]  r10:00000036 r9:93fec000 r8:80100204 r7:00000036 r6:00000009 r5:00000000
[   70.517291]  r4:045b7000
[   70.517299] ---[ end trace 0fa2dc075c66b864 ]---`

bibitocarlos, 2022-03-20:
I made it work.
I used the new stack with the ix219 module, switching back to the old stack have made it works.

closed 2022-03-20 by bibitocarlos