mpromonet/v4l2rtspserver #269: cant play http stream

bibitocarlos, 2022-03-20:
**Describe the bug**
I setup v4l2rtspserver.
`v4l2rtspserver -F 10 -H 1080 -W 1920 -P 8080 -S /dev/video0`

Playing rtsp with Vlc works, but i cant play http stream :+1: 
`vlc http://192.168.1.12:8080/ts.m3u8
VLC media player 3.0.17.3 Vetinari (revision 3.0.13-8-g41878ff4f2)
[000055ef9cf31ae0] main libvlc: Running vlc with the default interface. Use 'cvlc' to use vlc without interface.
[00007f085c001140] adaptive demux: Changing stream format Unknown -> TS
[00007f085c001140] adaptive demux error: Failed to create demuxer (nil) TS
uint DBusMenuExporterDBus::GetLayout(int, int, const QStringList&, DBusMenuLayoutItem&): Condition failed: menu
uint DBusMenuExporterDBus::GetLayout(int, int, const QStringList&, DBusMenuLayoutItem&): Condition failed: menu`

**Expected behavior**
To work

**Desktop (please complete the following information):**
raspberry pi os
latest

mpromonet, 2022-03-20:
Hi,

HLS works only with H264/H265 capture.

Best Regards,
Michel.

bibitocarlos, 2022-03-20:
Hi,

Do you know how to choose the h264 stream from the camera with v4l2rtspstream ?
`v4l2-ctl --list-formats
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'YU12' (Planar YUV 4:2:0)
        [1]: 'YUYV' (YUYV 4:2:2)
        [2]: 'RGB3' (24-bit RGB 8-8-8)
        [3]: 'JPEG' (JFIF JPEG, compressed)
        [4]: 'H264' (H.264, compressed)
        [5]: 'MJPG' (Motion-JPEG, compressed)
        [6]: 'YVYU' (YVYU 4:2:2)
        [7]: 'VYUY' (VYUY 4:2:2)
        [8]: 'UYVY' (UYVY 4:2:2)
        [9]: 'NV12' (Y/CbCr 4:2:0)
        [10]: 'BGR3' (24-bit BGR 8-8-8)
        [11]: 'YV12' (Planar YVU 4:2:0)
        [12]: 'NV21' (Y/CrCb 4:2:0)
        [13]: 'RX24' (32-bit XBGR 8-8-8-8)`

mpromonet, 2022-03-20:
Hi,

By default it use H264, then MJPEG, then JPEG. You can select it usign `-fH264`.

Best Regards,
Michel.

mpromonet, 2022-05-10:
Hi,

I close the issue, as I don't understand the point, you may open a new one with more details.

Best Regards,
Michel.

closed 2022-05-10 by mpromonet