mpromonet/v4l2rtspserver #270: RTSP over WLAN0

Tormund1985, 2022-03-21:
**Is your feature request related to a problem? Please describe.**
I have an RTSP server installed in a raspberry PI. streaming over internet throught nrok and connected by wifi with my smartphone. 

**Describe the solution you'd like**
I want to play the streaming of the raspberry with my smarphone in VLC.

**Describe alternatives you've considered**
Journalctl - f shows:
uri="rtsp://192.168.1.144:8554/unicast/track1"

**Additional context**
Thats URI is not correct because I want connect by Wifi, not LAN.

Can I change the interface?

Many thanks.

mpromonet, 2022-03-21:
Hi,

By default live555 choose interface sending a multicast packet and looking which interface is receiving it.
You can override this default choice using  `-I <wlan ip>` argument.

Best Regards,
Michel.

Tormund1985, 2022-03-22:
> Hi,
> 
> By default live555 choose interface sending a multicast packet and looking which interface is receiving it. You can override this default choice using `-I <wlan ip>` argument.
> 
> Best Regards, Michel.

HI again,

I test with "-I x.x.x.x", "-I x.x.x.x:8554" and -I "rtsp://x.x.x.x:8554" and no changes, follow with exit: 

[mar 22 08:00:18 sentinel v4l2rtspserver[26198]: Authorization: Digest username="admin", realm="LIVE555 Streaming Media", nonce="xxxxxxxxxxxxxx", uri="rtsp://192.168.1.144:8554/unicast/track1", response="xxxxxxxxxxxxxxxxxx"](url)

thanks in advance for your support!

mpromonet, 2022-05-10:
Hi,

It should be `-I x.x.x.x` 

Best Regards,
Michel.