mpromonet/v4l2rtspserver #271: Only Partial Image when using on Raspbian with Resolution set to 1920x1080

minestrone1994, 2022-03-23:
I use a RaspberryPi 3b+ with a standard v1.2 RaspberryPi Camera. I run Raspbian Bullseye, but changed it so it works with the legacy camera drivers. I installed and compiled your v4l2rtspserver as described in the readme, which worked flawless. 
I want to use this setup to stream my video feed to a local Homeassistant install - which works fine as long as I keep the resolution at 640x480. As soon as I set it to 1920x1080 it sort of zooms in and only shows the upper left part of the image (although in a higher resolution). How do I change this, so that it actually streams the full image in the correct resolution?
Any help highly regarded!

davebuk, 2022-05-19:
You could try 1920x1440. This is how I've been running this type of camera on motioneyeos.

y8s, 2022-08-25:
Different sensor modes on the pi camera have different crop factors. See this documentation:

https://picamera.readthedocs.io/en/release-1.13/fov.html#sensor-modes