mpromonet/v4l2rtspserver #276: Insert audio over rtsp

Tormund1985, 2022-05-05:
**Describe the bug**
I can't add audio over rtsp connection. It shows -help out

**To Reproduce**
./v4l2rtspserver -W 1280 -H 720 -F 25 /dev/video0,hw:1 -A 48000 -a S32_LE -p 8555

**Expected behavior**
When I insert any command concerning audio it shows the help output.
The video works, I can play it from VLC, but not the audio coming from a microphone.

**Screenshots**
`root@raspberrypi:/# arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
root@raspberrypi:/# `

`may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -v               : verbose
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -vv              : very verbose
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -Q <length>      : Number of frame queue  (default 5)
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -O <output>      : Copy captured frame to a file or a V4L2 device
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -b <webroot>     : path to webroot
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          RTSP/RTP options
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -I <addr>        : RTSP interface (default autodetect)
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -P <port>        : RTSP port (default 8554)
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -p <port>        : RTSP over HTTP port (default 8554)
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -U <user>:<pass> : RTSP user and password
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -R <realm>       : use md5 password 'md5(<username>:<realm>:<password>')
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -u <url>         : unicast url (default unicast)
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -m <url>         : multicast url (default multicast)
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -M <addr>        : multicast group:port (default is random_address:20000)
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -c               : don't repeat config (default repeat config before IDR frame)
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -t <timeout>     : RTCP expiration timeout in seconds (default 65)
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -S[<duration>]   : enable HLS & MPEG-DASH with segment duration  in seconds (default 2)
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -x <sslkeycert>  : enable RTSPS & SRTP
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          V4L2 options
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -r               : V4L2 capture using read interface (default use memory mapped buffers)
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -w               : V4L2 capture using write interface (default use memory mapped buffers)
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -B               : V4L2 capture using blocking mode (default use non-blocking mode)
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -s               : V4L2 capture using live555 mainloop (default use a reader thread)
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -f               : V4L2 capture using current capture format (-W,-H,-F are ignored)
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -f<format>       : V4L2 capture using format (-W,-H,-F are used)
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -W <width>       : V4L2 capture width (default 1920)
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -H <height>      : V4L2 capture height (default 1080)
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -F <fps>         : V4L2 capture framerate (default 10)
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          -G <w>x<h>[x<f>] : V4L2 capture format (default 1920x1080x10)
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          Devices :
may 05 11:48:56 raspberrypi v4l2rtspserver[1067]:          [V4L2 device][,ALSA device] : V4L2 capture device or/and ALSA capture device (default /dev/video0,/dev/video0)
may 05 11:48:56 raspberrypi systemd[1]: v4l2rtspserver.service: Succeeded.`

**Desktop (please complete the following information):**
 - OS: Raspian Buster

**Smartphone (please complete the following information):**
 - Device: Rasperrry Pi Zero 2W

**Additional context**
Audio works fine with ffmpeg command:

`ffmpeg -ar 44100 -ac 1 -f alsa -i plughw:1,0 -f wav -listen 1 tcp://0.0.0.0:5002`

mpromonet, 2022-05-05:
Hi,

As you can see in help, there is no audio options. This means audio support was not available. You should rebuilt with audio lib installed. Running `make VERBOSE=1` will show you if the compilation flag `HAVE_ALSA` is set.

Best regards
Michel

Tormund1985, 2022-05-06:
> Hi,
> 
> You may try `-A48000 -C1 -aS16_LE` that seems better correspond to your capture device.
> 
> Best regards Michel

HI Michel!

Now its working with next command:

`.../dev/video0,hw:1,0 -A48000 -C1 -aS16_LE `

I can't hear because I was trying to reproduce the rtsp "ts", but with unicast its working.

Thanks you very much, I really appreciate your help.

closed 2022-06-26 by mpromonet