mpromonet/v4l2rtspserver #277: two streams with the same source at the same time

ItsAMeMario1985, 2022-05-07:
Hi there, 

is there a way to setup two streams with the same source at the same time? I've a raspberry pi and raspberry cam and want to use one rtsp stream in hq and a second one with lower resolution to send it to youtube.

Thx!

mpromonet, 2022-05-08:
Hi,

You might look to v4l2loopback it will allow to uncompress hq and recompress lq. Then you can stream both.

Best regards
Michel

shamlord, 2022-05-16:
Is there any way to do this with v4l2rtspserver and v4l2loopback so that both the high and low resolution streams are available on the same RTSP port number, but with different URLs?

For example:
rtsp://192.168.2.17:8554/unicast
rtsp://192.168.2.17:8554/unicast_low

This is the typical implementation of high and low resolution streams on IP cameras offering RTSP.

Reading through the v4l2loopback project, it only appears to create loopback devices rather than creating an RTSP stream itself.

mpromonet, 2022-05-17:
Hi,

If your camera have 2 v4l2 devices (one for each resolution), you do not need v4l2loopback. The proposition to use v4l2loopback was to uncompress the high quality and recompress to low quality to this virtual device.

Best Regards,
Michel.

shamlord, 2022-05-17:
Per this thread: https://ipcamtalk.com/threads/adding-camera-that-the-substream-are-on-a-different-ports.59539/

IP cameras that provide multiple RTSP streams _typically_ provide those streams on the same port, but use different URLs. For example, all of these streams would be offered by 1 RTSP server on the camera, on the same port:
```
rtsp://x.x.x.x:554/unicast
rtsp://x.x.x.x:554/unicast2
```

Is there a way to do this with v4l2rtspserver?  Two streams with audio, on the same port but at different URLs?  Perhaps like this?
`v4l2rtspserver -u unicast /dev/video0,/dev/video0 -u unicast2 /dev/video1,/dev/video1`

mpromonet, 2022-05-17:
Hi,

When you add multiple device, the url become rtsp://.../video0/unicast rtsp://.../video1/unicast....

Best Regards,
Michel.

ItsAMeMario1985, 2022-05-17:
> You might look to v4l2loopback it will allow to uncompress hq and recompress lq. Then you can stream both.

Thx @mpromonet I'll take a look on v4l2loopback.

Best
Mario

gtxaspec, 2022-05-18:
@mpromonet is there any performance benefit to running two streams on one process? compares to two individual instances?

gtxaspec, 2022-06-12:
Hi @mpromonet 

Running the command: `v4l2rtspserver -C 1 -a S16_LE -U admin:xxx -P 8554 /dev/video1,hw:Loopback,0 /dev/video2,hw:Loopback,1`

produces errors:
```
log level:500

[NOTICE] /root/CLEAN/v4l2rtspserver/main.cpp:297
	Version: 0.3.3-4-ga43e094 live555 version:2022.04.26
[NOTICE] /root/CLEAN/v4l2rtspserver/src/V4l2RTSPServer.cpp:37
	Create V4L2 Source.../dev/video1
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:133
	driver:v4l2 loopback capabilities:85008003 mandatory:4000001
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:135
	/dev/video1 support output
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:136
	/dev/video1 support capture
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:138
	/dev/video1 support read/write
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:139
	/dev/video1 support streaming
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:225
	/dev/video1:H264 size:1920x1080 bufferSize:8294400
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:246
	fps:1/25
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:247
	nbBuffer:2
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:49
	Device /dev/video1
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:73
	Device /dev/video1 nb buffer:2
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:49
	VIDIOC_REQBUFS: Inappropriate ioctl for device
Device video1
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:141
	VIDIOC_STREAMOFF: Inappropriate ioctl for device
VIDIOC_REQBUFS: Inappropriate ioctl for device
Device video1
[WARN] /root/CLEAN/v4l2rtspserver/src/V4l2RTSPServer.cpp:53
	Cannot open output:video1
[NOTICE] /root/CLEAN/v4l2rtspserver/src/V4l2RTSPServer.cpp:62
	Create Source .../dev/video1
[NOTICE] /root/CLEAN/v4l2rtspserver/src/V4L2DeviceSource.cpp:96
	begin thread
[NOTICE] /root/CLEAN/v4l2rtspserver/src/V4l2RTSPServer.cpp:184
	Create ALSA Source...hw:Loopback,0
[NOTICE] /root/CLEAN/v4l2rtspserver/src/ALSACapture.cpp:49
	Open ALSA device: "hw:Loopback,0"
[NOTICE] /root/CLEAN/v4l2rtspserver/src/ALSACapture.cpp:117
	set sample format device: hw:Loopback,0 to:2 ok
[NOTICE] /root/CLEAN/v4l2rtspserver/src/ALSACapture.cpp:104
	ALSA device: "hw:Loopback,0" buffer_size:1048576 period_size:1024 rate:16000
[NOTICE] /root/CLEAN/v4l2rtspserver/inc/BaseServerMediaSubsession.h:49
	format:video/H264
[NOTICE] /root/CLEAN/v4l2rtspserver/inc/BaseServerMediaSubsession.h:49
	format:audio/L16/16000/1
[NOTICE] /root/CLEAN/v4l2rtspserver/inc/V4l2RTSPServer.h:80
	Play this stream using the URL "rtsp://x.x.x.x:8554/video1_unicast"
[NOTICE] /root/CLEAN/v4l2rtspserver/src/V4l2RTSPServer.cpp:37
	Create V4L2 Source.../dev/video2
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:133
	driver:v4l2 loopback capabilities:85008003 mandatory:4000001
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:135
	/dev/video2 support output
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:136
	/dev/video2 support capture
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:138
	/dev/video2 support read/write
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:139
	/dev/video2 support streaming
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:225
	/dev/video2:H264 size:640x360 bufferSize:921600
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:246
	fps:1/25
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2Device.cpp:247
	nbBuffer:2
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:49
	Device /dev/video2
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:73
	Device /dev/video2 nb buffer:2
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:49
	VIDIOC_REQBUFS: Inappropriate ioctl for device
Device video2
[NOTICE] /root/CLEAN/v4l2rtspserver/v4l2wrapper/src/V4l2MmapDevice.cpp:141
	VIDIOC_STREAMOFF: Inappropriate ioctl for device
VIDIOC_REQBUFS: Inappropriate ioctl for device
Device video2
[WARN] /root/CLEAN/v4l2rtspserver/src/V4l2RTSPServer.cpp:53
	Cannot open output:video2
[NOTICE] /root/CLEAN/v4l2rtspserver/src/V4l2RTSPServer.cpp:62
	Create Source .../dev/video2
[NOTICE] /root/CLEAN/v4l2rtspserver/src/V4L2DeviceSource.cpp:96
	begin thread
[NOTICE] /root/CLEAN/v4l2rtspserver/src/V4L2DeviceSource.cpp:96
	begin thread
[NOTICE] /root/CLEAN/v4l2rtspserver/src/V4l2RTSPServer.cpp:184
	Create ALSA Source...hw:Loopback,1
[NOTICE] /root/CLEAN/v4l2rtspserver/src/ALSACapture.cpp:49
	Open ALSA device: "hw:Loopback,1"
[NOTICE] /root/CLEAN/v4l2rtspserver/src/ALSACapture.cpp:117
	set sample format device: hw:Loopback,1 to:2 ok
[NOTICE] /root/CLEAN/v4l2rtspserver/src/ALSACapture.cpp:104
	ALSA device: "hw:Loopback,1" buffer_size:1048576 period_size:1024 rate:8000
[NOTICE] /root/CLEAN/v4l2rtspserver/inc/BaseServerMediaSubsession.h:49
	format:video/H264
[NOTICE] /root/CLEAN/v4l2rtspserver/inc/BaseServerMediaSubsession.h:49
	format:audio/L16/8000/1
[NOTICE] /root/CLEAN/v4l2rtspserver/inc/V4l2RTSPServer.h:80
	Play this stream using the URL "rtsp://xxx:8554/video2_unicast"
[NOTICE] /root/CLEAN/v4l2rtspserver/src/V4L2DeviceSource.cpp:96	
```

notably, the errors: 

```
	VIDIOC_REQBUFS: Inappropriate ioctl for device
	VIDIOC_STREAMOFF: Inappropriate ioctl for device
```

I dont see these errors when i do a single interface, or running two seperate v4l2rtspserver instances with different network ports.  These are v4l2loopback devices (video1 and video2)

gtxaspec, 2022-06-12:
heres an strace for reference
```
strace  /bin/v4l2rtspserver -C 1 -a S16_LE -U admin:xxx -P 8554 /dev/video1,hw:Loopback,0 /dev/video2,hw:Loopback,1
execve("/bin/v4l2rtspserver", ["/bin/v4l2rtspserver", "-C", "1", "-a", "S16_LE", "-U", "admin:xxx", "-P", "8554", "/dev/video1,hw:Loopback,0", "/dev/video2,hw:Loopback,1"], 0x7f819a78 /* 12 vars */) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|1<<MAP_HUGE_SHIFT, -1, 0) = 0x7735f000
stat("/etc/ld.so.cache", 0x7f8e32d8)    = -1 ENOENT (No such file or directory)
open("/lib/libasound.so.2", O_RDONLY) = 3
fstat(3, {st_mode=S_IFREG|0777, st_size=3051008, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|1<<MAP_HUGE_SHIFT, -1, 0) = 0x7735e000
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0\10\0\1\0\0\0\3006\1\0004\0\0\0"..., 4096) = 4096
mmap(NULL, 794624, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x77287000
mmap(0x77287000, 713524, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0x77287000
mmap(0x77345000, 14284, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0xae000) = 0x77345000
close(3)                                = 0
munmap(0x7735e000, 4096)                = 0
open("/lib/libm.so.0", O_RDONLY) = -1 ENOENT (No such file or directory)
open("/thirdlib/libm.so.0", O_RDONLY)   = -1 ENOENT (No such file or directory)
open("/system/lib/libm.so.0", O_RDONLY) = -1 ENOENT (No such file or directory)
open("/lib/libm.so.0", O_RDONLY)        = 3
fstat(3, {st_mode=S_IFREG|0755, st_size=48896, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|1<<MAP_HUGE_SHIFT, -1, 0) = 0x7735e000
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0\10\0\1\0\0\0\260\22\0\0004\0\0\0"..., 4096) = 4096
mmap(NULL, 110592, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7726c000
mmap(0x7726c000, 43416, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0x7726c000
mmap(0x77286000, 2816, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0xa000) = 0x77286000
close(3)                                = 0
munmap(0x7735e000, 4096)                = 0
open("/lib/libpthread.so.0", O_RDONLY) = -1 ENOENT (No such file or directory)
open("/thirdlib/libpthread.so.0", O_RDONLY) = -1 ENOENT (No such file or directory)
open("/system/lib/libpthread.so.0", O_RDONLY) = -1 ENOENT (No such file or directory)
open("/lib/libpthread.so.0", O_RDONLY)  = 3
fstat(3, {st_mode=S_IFREG|0755, st_size=84560, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|1<<MAP_HUGE_SHIFT, -1, 0) = 0x7735e000
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0\10\0\1\0\0\0\200?\0\0004\0\0\0"..., 4096) = 4096
mmap(NULL, 151552, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x77247000
mmap(0x77247000, 74664, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0x77247000
mmap(0x77269000, 2012, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x12000) = 0x77269000
mmap(0x7726a000, 6384, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7726a000
close(3)                                = 0
munmap(0x7735e000, 4096)                = 0
open("/lib/libc.so.0", O_RDONLY) = -1 ENOENT (No such file or directory)
open("/thirdlib/libc.so.0", O_RDONLY)   = -1 ENOENT (No such file or directory)
open("/system/lib/libc.so.0", O_RDONLY) = -1 ENOENT (No such file or directory)
open("/lib/libc.so.0", O_RDONLY)        = 3
fstat(3, {st_mode=S_IFREG|0755, st_size=747936, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|1<<MAP_HUGE_SHIFT, -1, 0) = 0x7735e000
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0\10\0\1\0\0\0 \264\0\0004\0\0\0"..., 4096) = 4096
mmap(NULL, 786432, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x77187000
mmap(0x77187000, 691264, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0x77187000
mmap(0x77240000, 4496, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0xa9000) = 0x77240000
mmap(0x77242000, 19628, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x77242000
close(3)                                = 0
munmap(0x7735e000, 4096)                = 0
stat("/lib/ld-uClibc.so.0", {st_mode=S_IFREG|0755, st_size=31692, ...}) = 0
open("/lib/libm.so.0", O_RDONLY) = -1 ENOENT (No such file or directory)
open("/thirdlib/libm.so.0", O_RDONLY)   = -1 ENOENT (No such file or directory)
open("/system/lib/libm.so.0", O_RDONLY) = -1 ENOENT (No such file or directory)
open("/lib/libm.so.0", O_RDONLY)        = 3
fstat(3, {st_mode=S_IFREG|0755, st_size=48896, ...}) = 0
close(3)                                = 0
open("/lib/libdl.so.0", O_RDONLY) = -1 ENOENT (No such file or directory)
open("/thirdlib/libdl.so.0", O_RDONLY)  = -1 ENOENT (No such file or directory)
open("/system/lib/libdl.so.0", O_RDONLY) = -1 ENOENT (No such file or directory)
open("/lib/libdl.so.0", O_RDONLY)       = 3
fstat(3, {st_mode=S_IFREG|0755, st_size=10928, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|1<<MAP_HUGE_SHIFT, -1, 0) = 0x7735e000
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0\10\0\1\0\0\0P\n\0\0004\0\0\0"..., 4096) = 4096
mmap(NULL, 77824, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x77174000
mmap(0x77174000, 9312, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0x77174000
mmap(0x77186000, 1400, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x2000) = 0x77186000
[... 1062 lines cut ...]
write(1, "NOTICE] /root/CLEAN/v4l2rtspserv"..., 60NOTICE] /root/CLEAN/v4l2rtspserver/src/ALSACapture.cpp:104
	) = 60
[... cut ...]
write(1, "format:audio/L16/8000/1\n[", 25format:audio/L16/8000/1
[) = 25
write(1, "NOTICE] /root/CLEAN/v4l2rtspserv"..., 60NOTICE] /root/CLEAN/v4l2rtspserver/inc/V4l2RTSPServer.h:80
	) = 60
rt_sigaction(SIGINT, {sa_handler=0x472df0, sa_mask=[INT], sa_flags=SA_RESTART}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 16) = 0
gettimeofday({tv_sec=1655020785, tv_usec=491793}, NULL) = 0
Play this stream using the URL "rtsp://x.x.x.x:8554/video2_unicast"
[NOTICE] /root/CLEAN/v4l2rtspserver/src/V4L2DeviceSource.cpp:96
```

mpromonet, 2022-06-12:
Hi gtxaspec,

There is clearly a bug managing output with multiple capture device.
I will fix this.

Best Regards,
Michel.

contactnat, 2022-08-22:
> Hi there,
> 
> is there a way to setup two streams with the same source at the same time? I've a raspberry pi and raspberry cam and want to use one rtsp stream in hq and a second one with lower resolution to send it to youtube.
> 
> Thx!

Hi - did you get yours working? I have the same/similar setup but cant figure this out, can you perhaps show your start command for v4l2rtspserver? 
mine looks like this: 
sudo v4l2rtspserver -W 1920 -H 1080 -F 5 -P 8554 /dev/video0

thanks

closed 2024-03-24 by mpromonet