mpromonet/v4l2rtspserver #279: make error

rockchung, 2022-05-11:
Hi all,

I have completed cmake 
But when I make the code, it showed the following error message
===========================
MJPEGVideoSource.cpp:(.text._ZN16MJPEGVideoSourceD2Ev[_ZN16MJPEGVideoSourceD5Ev]+0x34): undefined reference to `JPEGVideoSource::~JPEGVideoSource()'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/v4l2rtspserver.dir/build.make:92: v4l2rtspserver] Error 1
make[1]: *** [CMakeFiles/Makefile2:126: CMakeFiles/v4l2rtspserver.dir/all] Error 2
make: *** [Makefile:163: all] Error 2
===========================

I have check the code, and no ideas why it happened.
Any suggestions for that?

Thanks
Rock

mpromonet, 2022-05-11:
Hi Rock,

You should run make with `VERBOSE=1` and look to the link command. May be some mix between libraries ?

Best Regards,
Michel.

rockchung, 2022-05-12:
Hi Michel,

The following is the make result log.
[make2_log.txt](https://github.com/mpromonet/v4l2rtspserver/files/8674882/make2_log.txt)

It seems like link error. But I don't know how to solve this issue.
Do you or any have any suggestions?

best regards,
Rock

mpromonet, 2022-05-12:
Hi,

Probably the live555 libraries in /usr/local/lib are compiled with different options. Better to remove this and let cmake to download and compile it.

Best Regards,
Michel.

rockchung, 2022-05-13:
Hi Michel,

Thanks for your kindly support.
I will try it in the future.

The following is the reference instructions at internet.
I also re-executed at brand-new raspberry pi environment today.
And it seems it will happen the same error I encoutered.
==================
setup
======
Ref: http://c.wensheng.org/2017/05/18/stream-from-raspberrypi/

sudo apt-get install cmake liblog4cpp5-dev libv4l-dev
wget www.live555.com/liveMedia/public/live555-latest.tar.gz
tar xfz live555-latest.tar.gz
cd live/
./genMakefiles linux
make CPPFLAGS=-DALLOW_RTSP_SERVER_PORT_REUSE=1
sudo make install

cd
git clone https://github.com/mpromonet/v4l2rtspserver.git
cd v4l2rtspserver/
cmake .
make
sudo make install
==========================

Thanks again~
best regards,
Rock

mpromonet, 2022-05-13:
Hi Rock,

This is a not a really good idea to build live555 this way. Better to foolow [ReadMe](https://github.com/mpromonet/v4l2rtspserver#build) cmake will download live555 and build it with same compiler options than v4l2rtspserver. It will avoid mixing different options.

Best Regards,
Michel.

closed 2022-05-14 by mpromonet

rockchung, 2022-05-15:
Hi Michel,

I rebuild all and it works.
Thanks!

best regards,
Rock

commit d8edd42c (2022-05-14): fix #279: allow to use static live