mpromonet/v4l2rtspserver #280: [0.3.1] Cannot upgrade armv6l (wrong architecture)

wvdakker, 2022-05-15:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
1: use existing 0.3.0 install on a RPi 1B
2: Install version 0.3.1

**Expected behavior**
'Normal install'

**Screenshots**
Log:
`dpkg: error processing archive v4l2rtspserver-0.3.1-Linux-armv6l-.deb (--install):
 package architecture (armv6l) does not match system (armhf)`

By-the-way: armhf install but segfaulted.

**Desktop (please complete the following information):**
 - OS: 
`Linux 5.10.92+ #1514 Mon Jan 17 17:35:21 GMT 2022 armv6l GNU/Linux`

mpromonet, 2022-05-15:
Hi,

armhf is armv7, armv6 package architechture is wrongly build.

Best Regards,
Michel.

wvdakker, 2022-05-15:
Ok thanks. Is a new build be expected?

mpromonet, 2022-05-15:
I am checking it, the 0.3.2 should be ok.

wvdakker, 2022-05-15:
Great!

wvdakker, 2022-05-15:
It works with 0.3.3
Thanks

closed 2022-05-15 by wvdakker