mpromonet/v4l2rtspserver #281: H264 video mosaic is serious

Sleepingbug, 2022-05-17:
H264 video mosaic is serious, how to solve it ？

I use Logic C920 to test h264 video,
I found that if I moved the camera, the picture was better, but when I didn't move, the mosaic was bad.
I have the same issue with my Razer camera.
If I change to a MJPEG camera, there is no mosaic.
But I changed other software, the same is h264 encoding, there is no such problem.
How can I fix this?

mpromonet, 2022-05-17:
Hi,

I guess what you call mosaic is when your network drop frames ?
There is different possibilities, like reduce the resolution/framerate, use RTP/TCP.

Best Regards,
Michel.

mateuszdrab, 2025-08-09:
I just tried on a newly built Raspbian arm64 on a Pi3, latest build of [v4l2rtspserver](https://github.com/mpromonet/v4l2rtspserver) and when streaming a C920 all I get is:

[image]

It works slightly better in go2rtc, but not without issues unfortunately.

There's no network issues between the PC and the Pi, I think it might be the camera iteself being an issue.

mateuszdrab, 2025-08-10:
> Hi @mateuszdrab 
> 
> Probably I frame are lost, you could try RTP over TCP 
> 
> Best Regards 
> Michel 

That's what I was thinking.
 
Moved back to go2rtc and finally got it working although the picture becomes washy over time and I'm not sure if it's an rtsp problem or USB issue. 

I also switched back to raspbian from Ubuntu because on Ubuntu, the stream from the camera was dropping out seconds after starting, with some USB errors in dmesg.

It wasn't my first weird experience with Ubuntu on the Pi.