mpromonet/v4l2rtspserver #285: SPS/PPS missing on Describe

cedricve, 2022-06-26:
Hey @mpromonet amazing work here!

I'm using the joy4 library and noticed that the Describe method using your RTSP server is missing codec information (which is unexpected and is deviating from the RFC). Is there any particular reason for this?

https://github.com/kerberos-io/joy4/blob/master/format/rtsp/client.go#L664

mpromonet, 2022-06-26:
Hi @cedricve,

This is because your are capturing JPEG, the payload type 26.
Your device should be able to capture H264 frame, then you will get an H264 stream with SPS & PPS.

Best Regards,
Michel.

mpromonet, 2022-06-26:
Hi,

The approach is lightweight, just capture frames from v4l2 interface and publish through RTSP, encoding can be done between the real device and a v4l2loopback one.

Best Regards,
Michel.

closed 2022-07-30 by mpromonet