mpromonet/v4l2rtspserver #288: Would it be possible to send audio in mp3 format?

Regradert, 2022-09-06:
Hello. I'm having a problem streaming audio in the only formats that v4l2rtspserver allows. It turns out that the Shinobi program does not support PCM S16_BE, S16_LE, S32_BE or S32_LE audio. Besides, it is a very heavy audio and when it detects movement, the video clips weigh a lot.

I have also tried iSpy and it doesn't support PCM audio either.

The PCM audio format causes artifacts in the image when sending the transmission over Wi-Fi, since the video + audio is heavier.

Would it be possible to convert the audio from PCM to mp3 before sending it via rtsp? Is it very complicated to implement it?

I know that with FFmpeg it is very easy to transform the PCM into 128kb mp3, but it would have to be done before sending the transmission.

It would be a good idea if v4l2rtspserver could transform the PCM into mp3 so it doesn't affect the quality of the stream. It's a shame because it's one of the best systems I've tried.

mpromonet, 2022-09-06:
Hi,

This is not a new idea, there is a PR pending https://github.com/mpromonet/v4l2rtspserver/pull/247

Best Regards,
Michel.

Regradert, 2022-09-06:
Thanks. Hopefully soon we will have news of the implementation. It would be great as I have tried almost every system and this one is the best.