mpromonet/v4l2rtspserver #290: No valid mulitcast address with -M

wvdakker, 2022-10-11:
**Describe the bug**
Starting with the option -M should give a default multicast address.
However, it comes back with 255.255.255.255 as an multicast address.

**To Reproduce**
`v4l2rtspserver -M /dev/video0rt v4l2rtspserver -M /dev/video0`
outputs with 
`[NOTICE] /v4l2rtspserver/inc/V4l2RTSPServer.h:138
	RTP  address 255.255.255.255:20000
`

**Expected behavior**
I would expect a default multicast.
It would be nice if it generate a random IP address
where  nibble 2,3 and 4 are random chosen.

**Desktop (please complete the following information):**
Raspberry Pi 

**Additional context**
I also cannot see the multicast group in the IGMP router.
Changing to -M239.0.0.10 it works ok again.

mpromonet, 2022-10-12:
Hi @wvdakker 

This is becaus /dev/video0 is got for multicast address. Using `-m` without argument works as you expect.
I will modify the code to have similar behaviour, it will slightly change the behaviour of -M option, space will no more be accepted between option and value.

Best Regards,
Michel.

wvdakker, 2022-10-12:
@mpromonet  I can confirm it works in 0.3.7

closed 2022-10-12 by mpromonet