mpromonet/v4l2rtspserver #293: cross compile error : cannot find -lstdc++

mahdi2001h, 2022-10-30:
I want to compile this library for embedded linux system .
my compiler is from yocto

error:
```bash
[ 96%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/TSServerMediaSubsession.cpp.o
[ 96%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/UnicastServerMediaSubsession.cpp.o
[ 97%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/V4L2DeviceSource.cpp.o
[ 97%] Building CXX object CMakeFiles/libv4l2rtspserver.dir/src/V4l2RTSPServer.cpp.o
/home/user/v4l2rtspserver/src/V4L2DeviceSource.cpp: In member function ‘void V4L2DeviceSource::postFrame(char*, int, const timeval&)’:
/home/user/v4l2rtspserver/src/V4L2DeviceSource.cpp:229:8: warning: ignoring return value of ‘ssize_t write(int, const void*, size_t)’, declared with attribute warn_unused_result [-Wunused-result]
  229 |   write(m_outfd, frame, frameSize);
      |   ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 98%] Linking CXX static library liblibv4l2rtspserver.a
[ 98%] Built target libv4l2rtspserver
Scanning dependencies of target v4l2rtspserver
[ 98%] Building CXX object CMakeFiles/v4l2rtspserver.dir/main.cpp.o
[100%] Linking CXX executable v4l2rtspserver
/home/user/compiler/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/9.3.0/real-ld: cannot find -lstdc++
collect2: error: ld returned 1 exit status
CMakeFiles/v4l2rtspserver.dir/build.make:88: recipe for target 'v4l2rtspserver' failed
make[2]: *** [v4l2rtspserver] Error 1
```

I also checked my gcc patch in yocto and found this file:
`poky/meta/recipes-devtools/gcc/gcc-9.3/0005-optional-libstdc.patch`
```patch
From 435e45592e944018f33bff32f1202b0872a23141 Mon Sep 17 00:00:00 2001
From: Khem Raj <raj.khem@gmail.com>
Date: Fri, 29 Mar 2013 09:12:56 +0400
Subject: [PATCH 05/39] optional libstdc

gcc-runtime builds libstdc++ separately from gcc-cross-*. Its configure tests using g++
will not run correctly since by default the linker will try to link against libstdc++
which shouldn't exist yet. We need an option to disable -lstdc++
option whilst leaving -lc, -lgcc and other automatic library dependencies added by gcc
driver. This patch adds such an option which only disables the -lstdc++.

A "standard" gcc build uses xgcc and hence avoids this. We should ask upstream how to
do this officially, the likely answer is don't build libstdc++ separately.

RP 29/6/10

Upstream-Status: Inappropriate [embedded specific]
---
 gcc/c-family/c.opt  |  4 ++++
 gcc/cp/g++spec.c    |  1 +
 gcc/doc/invoke.texi | 32 +++++++++++++++++++++++++++++++-
 gcc/gcc.c           |  1 +
 4 files changed, 37 insertions(+), 1 deletion(-)

diff --git a/gcc/c-family/c.opt b/gcc/c-family/c.opt
index 916cc67b453..41619fa591c 100644
--- a/gcc/c-family/c.opt
+++ b/gcc/c-family/c.opt
@@ -1952,6 +1952,10 @@ nostdinc++
 C++ ObjC++
 Do not search standard system include directories for C++.
 
+nostdlib++
+Driver
+Do not link standard C++ runtime library
+
 o
 C ObjC C++ ObjC++ Joined Separate
 ; Documented in common.opt
diff --git a/gcc/cp/g++spec.c b/gcc/cp/g++spec.c
index 6c4574a837d..0e2657f00ee 100644
--- a/gcc/cp/g++spec.c
+++ b/gcc/cp/g++spec.c
@@ -137,6 +137,7 @@ lang_specific_driver (struct cl_decoded_option **in_decoded_options,
       switch (decoded_options[i].opt_index)
 	{
 	case OPT_nostdlib:
+	case OPT_nostdlib__:
 	case OPT_nodefaultlibs:
 	  library = -1;
 	  break;
diff --git a/gcc/doc/invoke.texi b/gcc/doc/invoke.texi
index c3d3d51a28f..2f7ffe456c3 100644
--- a/gcc/doc/invoke.texi
+++ b/gcc/doc/invoke.texi
@@ -228,6 +228,9 @@ in the following sections.
 -fno-weak  -nostdinc++ @gol
 -fvisibility-inlines-hidden @gol
 -fvisibility-ms-compat @gol
+-fvtable-verify=@r{[}std@r{|}preinit@r{|}none@r{]} @gol
+-fvtv-counts -fvtv-debug @gol
+-nostdlib++ @gol
 -fext-numeric-literals @gol
 -Wabi=@var{n}  -Wabi-tag  -Wconversion-null  -Wctor-dtor-privacy @gol
 -Wdelete-non-virtual-dtor  -Wdeprecated-copy  -Wdeprecated-copy-dtor @gol
@@ -539,7 +542,7 @@ Objective-C and Objective-C++ Dialects}.
 -pie  -pthread  -r  -rdynamic @gol
 -s  -static  -static-pie  -static-libgcc  -static-libstdc++ @gol
 -static-libasan  -static-libtsan  -static-liblsan  -static-libubsan @gol
--shared  -shared-libgcc  -symbolic @gol
+-shared  -shared-libgcc  -symbolic -nostdlib++ @gol
 -T @var{script}  -Wl,@var{option}  -Xlinker @var{option} @gol
 -u @var{symbol}  -z @var{keyword}}
 
@@ -13321,6 +13324,33 @@ Specify that the program entry point is @var{entry}.  The argument is
 interpreted by the linker; the GNU linker accepts either a symbol name
 or an address.
 
+@item -nostdlib++
+@opindex nostdlib++
+Do not use the standard system C++ runtime libraries when linking.
+Only the libraries you specify will be passed to the linker.
+
+@cindex @option{-lgcc}, use with @option{-nostdlib}
+@cindex @option{-nostdlib} and unresolved references
+@cindex unresolved references and @option{-nostdlib}
+@cindex @option{-lgcc}, use with @option{-nodefaultlibs}
+@cindex @option{-nodefaultlibs} and unresolved references
+@cindex unresolved references and @option{-nodefaultlibs}
+One of the standard libraries bypassed by @option{-nostdlib} and
+@option{-nodefaultlibs} is @file{libgcc.a}, a library of internal subroutines
+which GCC uses to overcome shortcomings of particular machines, or special
+needs for some languages.
+(@xref{Interface,,Interfacing to GCC Output,gccint,GNU Compiler
+Collection (GCC) Internals},
+for more discussion of @file{libgcc.a}.)
+In most cases, you need @file{libgcc.a} even when you want to avoid
+other standard libraries.  In other words, when you specify @option{-nostdlib}
+or @option{-nodefaultlibs} you should usually specify @option{-lgcc} as well.
+This ensures that you have no unresolved references to internal GCC
+library subroutines.
+(An example of such an internal subroutine is @code{__main}, used to ensure C++
+constructors are called; @pxref{Collect2,,@code{collect2}, gccint,
+GNU Compiler Collection (GCC) Internals}.)
+
 @item -pie
 @opindex pie
 Produce a dynamically linked position independent executable on targets
diff --git a/gcc/gcc.c b/gcc/gcc.c
index a2601a6bb06..cd6c6fc95db 100644
--- a/gcc/gcc.c
+++ b/gcc/gcc.c
@@ -1052,6 +1052,7 @@ proper position among the other output files.  */
     %(mflib) " STACK_SPLIT_SPEC "\
     %{fprofile-arcs|fprofile-generate*|coverage:-lgcov} " SANITIZER_SPEC " \
     %{!nostdlib:%{!r:%{!nodefaultlibs:%(link_ssp) %(link_gcc_c_sequence)}}}\
+    %{!nostdlib++:}\
     %{!nostdlib:%{!r:%{!nostartfiles:%E}}} %{T*}  \n%(post_link) }}}}}}"
 #endif
 
-- 
2.25.1

```

mpromonet, 2022-10-30:
Hi,

I made a similar change for openwrt some weeks ago.
DId you try with cmake flag `STATICSTDCPP=OFF` ?

Best Regards,
Michel.

mahdi2001h, 2022-10-30:
thank you
It worked :)

I changed file CmakeLists.txt line 9:
```bash
set(STATICSTDCPP ON CACHE BOOL "use gcc static lib if available")
```
to 
```bash
set(STATICSTDCPP OFF CACHE BOOL "use gcc static lib if available")
```

closed 2022-10-30 by mahdi2001h

mpromonet, 2022-10-30:
Hi,

You can also without modifying CMakeLists.txt, just running `cmake -DSTATICSTDCPP=OFF .`

Best Regards,
Michel.