mpromonet/v4l2rtspserver #294: Streaming using HLS

KirillYatsenko, 2022-11-20:
Hello,

First of all thanks for the great project,

The rtsp connection works and I can connect to the stream using vlc like this: **"vlc rtsp://192.168.0.3:8554/unicast"**

However after running **"v4l2rtspserver -S"** I can't connect to the stream with **"vlc http://192.168.0.3:8554/ts.m3u8 "**

I'm getting the following output:

[00007fc318001610] access stream error: HTTP 405 error
[00007fc318001610] http stream error: error: HTTP/1.0 405 Method Not Allowed

Maybe the solution is simple, and I'm missing something.

Best Regards

mpromonet, 2022-11-20:
Hi @KirillYatsenko, 

HLS & mpeg# are working only with H264 or H265, are you using one of these capture format ?

Best Regards,
Michel.

KirillYatsenko, 2022-11-20:
Thanks for the answer,

My camera doesn't support H264/H265,
So I'm need to use some software encoding as you described in README.

closed 2022-11-20 by KirillYatsenko