mpromonet/v4l2rtspserver #297: VLC can't connect to server, no errors in log

Drizzt321, 2023-01-10:
**Describe the bug**
I'm trying to run v4l2rtspserver to use my RPi Zero W as a small security camera. Got the latest raspbian, updated all packages to latest, then installed v4l2rtspserver from the git repo. Was initially having some trouble getting it started, then found the `LD_PRELOAD` tip on https://github.com/mpromonet/v4l2rtspserver/issues/283 for the new libcamera v4l2-compat support. 

I started it up using `LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/v4l2-compat.so v4l2rtspserver /dev/video0` as root (just to get initial settings right, then run as daemon as regular user). 

Starts up with this output

```
log level:500

[NOTICE] /root/v4l2rtspserver/v4l2rtspserver/main.cpp:269
        Version: 0.3.7-19-g9029be3 live555 version:2022.12.01
[NOTICE] /root/v4l2rtspserver/v4l2rtspserver/src/V4l2RTSPServer.cpp:37
        Create V4L2 Source.../dev/video0[0:03:10.497192926] [607]  INFO Camera camera_manager.cpp:299 libcamera v0.0.2+55-5df5b72c
[0:03:10.585622458] [608]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0

[NOTICE] /root/v4l2rtspserver/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:133
        driver:libcamera capabilities:84200001 mandatory:4000001
[NOTICE] /root/v4l2rtspserver/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:136
        /dev/video0 support capture
[NOTICE] /root/v4l2rtspserver/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:139
        /dev/video0 support streaming[0:03:10.587015220] [607]  WARN V4L2 v4l2_pixelformat.cpp:312 Unsupported V4L2 pixel format H264
[0:03:10.587700064] [607]  WARN Formats formats.cpp:838 Unsupported pixel format 0x00000000
[0:03:10.588187982] [607]  INFO Camera camera.cpp:1026 configuring streams: (0) 800x600-NV12
[0:03:10.588775430] [608]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC

[ERROR] /root/v4l2rtspserver/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:212
        /dev/video0: Cannot set pixelformat to:H264 format is:NV12[0:03:10.592964810] [607]  INFO Camera camera.cpp:1026 configuring streams: (0) 800x600-NV12
[0:03:10.593863405] [608]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC

[ERROR] /root/v4l2rtspserver/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:212
        /dev/video0: Cannot set pixelformat to:MJPG format is:NV12[0:03:10.596960388] [607]  INFO Camera camera.cpp:1026 configuring streams: (0) 800x600-NV12
[0:03:10.597838358] [608]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC

[ERROR] /root/v4l2rtspserver/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:212
        /dev/video0: Cannot set pixelformat to:JPEG format is:NV12[0:03:10.601019143] [607]  INFO Camera camera.cpp:1026 configuring streams: (0) 800x600-NV12
[0:03:10.601709039] [608]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC

[NOTICE] /root/v4l2rtspserver/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:225
        /dev/video0:NV12 size:800x600 bufferSize:720000
[WARN] /root/v4l2rtspserver/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:243
        Cannot set param for device:/dev/video0 Inappropriate ioctl for device
[NOTICE] /root/v4l2rtspserver/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:246
        fps:1/25
[NOTICE] /root/v4l2rtspserver/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:247
        nbBuffer:0
[NOTICE] /root/v4l2rtspserver/v4l2rtspserver/libv4l2cpp/src/V4l2MmapDevice.cpp:49
        Device /dev/video0[0:03:10.604888887] [607]  INFO Camera camera.cpp:1026 configuring streams: (0) 800x600-NV12
[0:03:10.605762325] [608]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC

[NOTICE] /root/v4l2rtspserver/v4l2rtspserver/libv4l2cpp/src/V4l2MmapDevice.cpp:73
        Device /dev/video0 nb buffer:10
[NOTICE] /root/v4l2rtspserver/v4l2rtspserver/src/V4l2RTSPServer.cpp:62
        Create Source .../dev/video0
[NOTICE] /root/v4l2rtspserver/v4l2rtspserver/src/V4L2DeviceSource.cpp:96
        begin thread
[NOTICE] /root/v4l2rtspserver/v4l2rtspserver/inc/BaseServerMediaSubsession.h:49
        format:video/RAW
[NOTICE] /root/v4l2rtspserver/v4l2rtspserver/inc/V4l2RTSPServer.h:237
```
I then try and connect from VLC on my desktop to `rtsp://192.168.10.70:8554/unicast`, and get an error in VLC

```
Your input can't be opened:
VLC is unable to open the MRL 'rtsp://192.168.10.70:8554/unicast'. Check the log for details.
```

On the Pi, I get this output 
```
handleCmd_SETUP:SETUP rtsp://192.168.10.70:8554/unicast RTSP/1.0
CSeq: 0
Transport: RTP/AVP;unicast;client_port=9284-9285
```

Can't see anything specific in the output, any ideas? Any tips on directions I could go in to try and figure out the issue?

mpromonet, 2023-01-10:
Hi,

Last time I tried VLC doesnot support NV12 format. You could use ffplay that support it.

Best Regards,
Michel.

Drizzt321, 2023-01-10:
Hm, so how do I change the format? I've tried using the `-f` option, but that doesn't seem to work. Would using the v4l2loopback and having the loopback provide the correct h264/whatever format work, assuming I use the `-f` with no options to tell v4l2rtspserver to use whatever the current v4l2 format is set to?

mpromonet, 2023-01-14:
Hi,

-f options has 2 usage without argument (it keep actual format) with argument (for instance -fH264, it set the format, and if format is not supported by device, it fails)

Best Regards,
Michel.

azsde, 2023-01-20:
I have the same issue as @Drizzt321 , I used [libcamerify](https://git.linuxtv.org/libcamera.git/tree/src/v4l2/libcamerify.in) instead of the LD_PRELOAD method, but I can't get the stream to be played neither by VLC (which is normal according to your previous answer) nor by ffplay.

```
azsde@azsdepi:~ $ libcamerify -d v4l2rtspserver -v
log level:600

[NOTICE] /home/azsde/v4l2rtspserver/main.cpp:269
	Version: 0.3.7-19-g9029be3 live555 version:2023.01.19
[NOTICE] /home/azsde/v4l2rtspserver/src/V4l2RTSPServer.cpp:37
	Create V4L2 Source.../dev/video0[0:25:35.077752681] [909]  INFO Camera camera_manager.cpp:299 libcamera v0.0.2+55-5df5b72c
[0:25:35.295452294] [910]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media3 and ISP device /dev/media0
[0:25:35.295979480] [909] DEBUG V4L2Compat v4l2_compat_manager.cpp:79 Started camera manager
[0:25:35.296492083] [909] DEBUG V4L2Compat v4l2_camera_proxy.cpp:50 [] open()
[0:25:35.297818486] [909] DEBUG V4L2Compat v4l2_compat_manager.cpp:163 Opened /dev/video0 -> fd 5
[0:25:35.297937339] [909] DEBUG V4L2Compat v4l2_camera_proxy.cpp:251 [/dev/video0 (fd:5)] vidioc_querycap()

[NOTICE] /home/azsde/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:133
	driver:libcamera capabilities:84200001 mandatory:4000001
[NOTICE] /home/azsde/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:136
	/dev/video0 support capture
[NOTICE] /home/azsde/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:139
	/dev/video0 support streaming[0:25:35.298334266] [909] DEBUG V4L2Compat v4l2_camera_proxy.cpp:307 [/dev/video0 (fd:5)] vidioc_g_fmt()
[0:25:35.298462963] [909] DEBUG V4L2Compat v4l2_camera_proxy.cpp:307 [/dev/video0 (fd:5)] vidioc_g_fmt()
[0:25:35.298661921] [909] DEBUG V4L2Compat v4l2_camera_proxy.cpp:351 [/dev/video0 (fd:5)] vidioc_s_fmt()
[0:25:35.298750567] [909]  WARN V4L2 v4l2_pixelformat.cpp:312 Unsupported V4L2 pixel format H264
[0:25:35.300440042] [909]  WARN Formats formats.cpp:838 Unsupported pixel format 0x00000000
[0:25:35.301179467] [909] DEBUG V4L2Compat v4l2_camera.cpp:127 Validated configuration is: 800x600-NV12
[0:25:35.301871236] [909]  INFO Camera camera.cpp:1026 configuring streams: (0) 800x600-NV12
[0:25:35.302844515] [910]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected unicam format: 1536x864-pBAA

[ERROR] /home/azsde/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:212
	/dev/video0: Cannot set pixelformat to:H264 format is:NV12[0:25:35.308144710] [909] DEBUG V4L2Compat v4l2_camera_proxy.cpp:307 [/dev/video0 (fd:5)] vidioc_g_fmt()
[0:25:35.308305699] [909] DEBUG V4L2Compat v4l2_camera_proxy.cpp:351 [/dev/video0 (fd:5)] vidioc_s_fmt()
[0:25:35.310449183] [909] DEBUG V4L2Compat v4l2_camera.cpp:127 Validated configuration is: 800x600-NV12
[0:25:35.311084494] [909]  INFO Camera camera.cpp:1026 configuring streams: (0) 800x600-NV12
[0:25:35.312114596] [910]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected unicam format: 1536x864-pBAA

[ERROR] /home/azsde/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:212
	/dev/video0: Cannot set pixelformat to:MJPG format is:NV12[0:25:35.315462244] [909] DEBUG V4L2Compat v4l2_camera_proxy.cpp:307 [/dev/video0 (fd:5)] vidioc_g_fmt()
[0:25:35.315631774] [909] DEBUG V4L2Compat v4l2_camera_proxy.cpp:351 [/dev/video0 (fd:5)] vidioc_s_fmt()
[0:25:35.317980050] [909] DEBUG V4L2Compat v4l2_camera.cpp:127 Validated configuration is: 800x600-NV12
[0:25:35.318596611] [909]  INFO Camera camera.cpp:1026 configuring streams: (0) 800x600-NV12
[0:25:35.319543379] [910]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected unicam format: 1536x864-pBAA

[ERROR] /home/azsde/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:212
	/dev/video0: Cannot set pixelformat to:JPEG format is:NV12[0:25:35.323085089] [909] DEBUG V4L2Compat v4l2_camera_proxy.cpp:307 [/dev/video0 (fd:5)] vidioc_g_fmt()
[0:25:35.323240349] [909] DEBUG V4L2Compat v4l2_camera_proxy.cpp:351 [/dev/video0 (fd:5)] vidioc_s_fmt()
[0:25:35.325589041] [909] DEBUG V4L2Compat v4l2_camera.cpp:127 Validated configuration is: 800x600-NV12
[0:25:35.326211592] [909]  INFO Camera camera.cpp:1026 configuring streams: (0) 800x600-NV12
[0:25:35.327151798] [910]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected unicam format: 1536x864-pBAA

[NOTICE] /home/azsde/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:225
	/dev/video0:NV12 size:800x600 bufferSize:720000[0:25:35.330701737] [909] DEBUG V4L2Compat v4l2_camera_proxy.cpp:307 [/dev/video0 (fd:5)] vidioc_g_fmt()

[WARN] /home/azsde/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:243
	Cannot set param for device:/dev/video0 Inappropriate ioctl for device
[NOTICE] /home/azsde/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:246
	fps:1/25
[NOTICE] /home/azsde/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:247
	nbBuffer:0
[NOTICE] /home/azsde/v4l2rtspserver/libv4l2cpp/src/V4l2MmapDevice.cpp:49
	Device /dev/video0[0:25:35.331294912] [909] DEBUG V4L2Compat v4l2_camera_proxy.cpp:478 [/dev/video0 (fd:5)] vidioc_reqbufs()
[0:25:35.331425797] [909] DEBUG V4L2Compat v4l2_camera_proxy.cpp:485 10 buffers requested 
[0:25:35.332222983] [909] DEBUG V4L2Compat v4l2_camera.cpp:127 Validated configuration is: 800x600-NV12
[... cut ...]
  libavformat    58. 76.100 / 58. 76.100
  libavdevice    58. 13.100 / 58. 13.100
  libavfilter     7.110.100 /  7.110.100
  libswscale      5.  9.100 /  5.  9.100
  libswresample   3.  9.100 /  3.  9.100
  libpostproc    55.  9.100 / 55.  9.100
[bitpacked @ 0x7ff238006480] The decoder 'bitpacked' is experimental but experimental codecs are not enabled, add '-strict -2' if you want to use it.
[rtsp @ 0x7ff238000cc0] Failed to open codec in avformat_find_stream_info
Floating point exception (core dumped)

```

azsde, 2023-01-23:
Hi Michel,

There seems to be some kind of "race condition" to get things somewhat working.

Here's the video feed I struggled to get:

[image]

My guess is that the format provided by the camera is not compatible ?

azsde, 2023-01-25:
Update: going back to the V2 camera has the same behavior, I'm not quite sure what is going wrong here.