mpromonet/v4l2rtspserver #298: Video fmtp not set

baudneo, 2023-01-11:
Hi, 

I use wz mini hacks which uses this software to stream rtsp.

The main 1080p stream does not include the video fmtp param which causes janus to error out.

DESCRIBE response:

mpromonet, 2023-01-14:
Hi,

I don't understand the point, the logs you show have a profile-level-id.

Best Regards,
Michel.

baudneo, 2023-01-15:
That was the wrong logs here are the logs of the one that fails. The first logs I posted are the substream, these following logs are the mainstream.

```
CSeq: 1
Date: Sun, Aug 07 2022 03:44:20 GMT
WWW-Authenticate: Digest realm="LIVE555 Streaming Media", nonce="2431aaa7fb9fa620e8665597fd90b059"

RTSP/1.0 200 OK
CSeq: 2
Date: Sun, Aug 07 2022 03:44:20 GMT
Content-Base: rtsp://10.0.1.67:8554/video1_unicast/
Content-Type: application/sdp
Content-Length: 563

v=0
o=- 1659831422316587 1 IN IP4 10.0.1.67
s=LIVE555 Streaming Media v2022.04.26
i=LIVE555 Streaming Media v2022.04.26
t=0 0
a=tool:LIVE555 Streaming Media v2022.04.26
a=type:broadcast
a=control:*
a=range:npt=now-
a=x-qt-text-nam:LIVE555 Streaming Media v2022.04.26
a=x-qt-text-inf:LIVE555 Streaming Media v2022.04.26
m=video 0 RTP/AVP 96
c=IN IP4 0.0.0.0
b=AS:500
a=rtpmap:96 H264/90000
a=fmtp:96
a=x-dimensions:1920,1080
a=control:track1
m=audio 0 RTP/AVP 97
c=IN IP4 0.0.0.0
b=AS:500
a=rtpmap:97 L16/16000
a=fmtp:97
a=control:track2

[janus-ZM] [Sat Aug  6 21:44:22 2022] [ERR] [plugins/janus_streaming.c:janus_streaming_rtsp_parse_sdp:6349] [2] cannot parse video fmtp...
[janus-ZM] [Sat Aug  6 21:44:22 2022] [ERR] [plugins/janus_streaming.c:janus_streaming_process_synchronous_request:3254] Error creating 'rtsp' stream...
```

So the substream is successful becvause it has profile-level-id set but in the mainstream it fails due to no profile-level-id

mpromonet, 2023-01-15:
Hi,

This is set https://github.com/mpromonet/v4l2rtspserver/blob/eaeda29aec14048ceaf11c2159b978f64cba4bf4/src/ServerMediaSubsession.cpp#L124
Using data are stored in https://github.com/mpromonet/v4l2rtspserver/blob/eaeda29aec14048ceaf11c2159b978f64cba4bf4/src/H264_V4l2DeviceSource.cpp#L63.
 
So it seems sps/pps are not found in the capture. Looking to the detail log may helps you to understand if it comes from your device, or an error in parsing.

Best Regards,
Michel.

baudneo, 2023-01-28:
Hi, 
So the setup using v4lLoopback -> v4lrtspserver -> rtsp client.

I have some log output using -vv in a gist. [Here is where PPS and SPS show up for the first time in the logs](https://gist.github.com/baudneo/bf66026b3fb058265b0bda057d755f19#file-v4lrtsp-log-L184). SPS and PPS being sent means profile-level-id and sprop should be set?

closed 2023-12-14 by baudneo