mpromonet/v4l2rtspserver #309: -O argument with v4l2 device shows just first frame

CyberKachok, 2023-04-04:
**Describe the bug**
After creating device using v4l2loopback i am trying to run v4l2server using v4l2rtspserver -O /dev/video5 -W 1920 -H 1080 -F 30 -P 7447 /dev/video0, rtsp stream works but when i am trying to capture /dev/video5 i get just first frame.

**Desktop (please complete the following information):**
 - OS: [Ubuntu Mate]

mpromonet, 2023-04-29:
Hi,

The -O option was more for debug purpose, I will probably remove it.

Best Regards,
Michel.

CyberKachok, 2023-04-29:
Hi, @mpromonet, I was able to use it, it is very usefull if you want to stream with minimum delay and get frames for another program on device, additional reencoding from one virtual camera to another using ffmpeg helped me.

mpromonet, 2023-04-29:
Hi,

Using v4l2loopback as a source, allow you to connect multiple consumer.

Best Regards,
Michel.

CyberKachok, 2023-04-29:
@mpromonet Yeah, but in my case using v4l2loopback as a source for rtspserver multiplied the delay

closed 2023-04-29 by CyberKachok