mpromonet/v4l2rtspserver #318: v4l2rtspserver server stops streaming when grabbing a still from the same video device (using v4l2loopback and v4l2copy)

dragulesq, 2023-06-20:
Hello and thanks for your wonderful software.
I've been using for a few years now with several raspi zeroes and it's been solid.

Yesterday, I ran into an issue where **v4l2rtspserver** stops streaming to already connected clients as well as new connections. VLC and ffplay both report timeouts. This happens when I try to use **ffmpeg** to capture one still frame from the same video device that **v4l2rtspserver** is using.

I am using **v4l2loopback** and **v4l2copy** from `/dev/video0` to `/dev/video2` as I would like to have two consumers of `/dev/video2`: **v4l2rtspserver** always and **ffmpeg** occasionally. I am running this on a bit of an exotic hardware as a proof of concept, Seagate Dockstar with an old Logitech C900 USB Camera that supports MJPG natively.
```
$ uname -a
Linux tv2 6.0.7-kirkwood-tld-1 #1.0 PREEMPT Fri Nov 4 15:08:17 PDT 2022 armv5tel GNU/Linux

$ dmesg | grep usb | grep UVC
[   25.966606] usb 1-1.3: Found UVC 1.00 device <unnamed> (046d:0990)
[   26.065923] input: UVC Camera (046d:0990) as /devices/platform/ocp@f1000000/f1050000.ehci/usb1/1-1/1-1.3/1-1.3:1.0/input/input0

$ v4l2-ctl --list-formats -d 0 /dev/video0
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'MJPG' (Motion-JPEG, compressed)
        [1]: 'YUYV' (YUYV 4:2:2)
```

Here is my software setup:

```(shell)
$ sudo modprobe v4l2loopback video_nr=2
$ v4l2-ctl -d /dev/video0 --set-fmt-video=width=960,height=720,pixelformat=MJPG --set-parm=10
$ nohup v4l2copy /dev/video0 /dev/video2 &> v4l2copy.log &
$ nohup v4l2rtspserver -fMJPG -F 10 -W 960 -H 720 /dev/video2,0 &> v4l2rtspserver.log &
```

Here is the ffmpeg command I use to grab a still frame from `/dev/video2` :
```
ffmpeg -v 32 -hide_banner -f video4linux2 -input_format mjpeg -i /dev/video2 -vframes 1 output%3d.jpg
Input #0, video4linux2,v4l2, from '/dev/video2':
  Duration: N/A, start: 356418.543514, bitrate: N/A
  Stream #0:0: Video: mjpeg (Baseline), yuvj422p(pc, bt470bg/unknown/unknown), 960x720, 10 fps, 10 tbr, 1000k tbn
Stream mapping:
  Stream #0:0 -> #0:0 (mjpeg (native) -> mjpeg (native))
Press [q] to stop, [?] for help
Output #0, image2, to 'output%3d.jpg':
  Metadata:
    encoder         : Lavf60.3.100
  Stream #0:0: Video: mjpeg, yuvj422p(pc, bt470bg/unknown/unknown, progressive), 960x720, q=2-31, 200 kb/s, 10 fps, 10 tbn
    Metadata:
      encoder         : Lavc60.3.100 mjpeg
    Side data:
      cpb: bitrate max/min/avg: 0/0/200000 buffer size: 0 vbv_delay: N/A
frame=    1 fps=0.0 q=5.7 Lsize=N/A time=00:00:00.00 bitrate=N/A speed=   0x
video:45kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown
```

Here is the v4lrtspserver verbose output, there is the client connecting in the middle and then the error at the end:
```
$ v4l2rtspserver -vv -fMJPG -F 10 -W 960 -H 720 /dev/video2 &> v4l2rtspserver.log &
[NOTICE] /home/username/v4l2rtspserver/main.cpp:269
        Version: 0.3.7-19-g9029be3 live555 version:2022.10.01
[NOTICE] /home/username/v4l2rtspserver/src/V4l2RTSPServer.cpp:37
        Create V4L2 Source.../dev/video2
[NOTICE] /home/username/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:133
        driver:v4l2 loopback capabilities:85200003 mandatory:4000001
[NOTICE] /home/username/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:135
        /dev/video2 support output
[NOTICE] /home/username/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:136
        /dev/video2 support capture
[NOTICE] /home/username/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:138
        /dev/video2 support read/write
[NOTICE] /home/username/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:139
        /dev/video2 support streaming
[DEBUG] /home/username/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:71
        /dev/video2:MJPG size:960x720 bufferSize:2764800
[NOTICE] /home/username/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:225
        /dev/video2:MJPG size:960x720 bufferSize:2764800
[DEBUG] /home/username/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:71
        /dev/video2:MJPG size:960x720 bufferSize:2764800
[NOTICE] /home/username/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:246
        fps:1/10
[NOTICE] /home/username/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:247
        nbBuffer:2
[NOTICE] /home/username/v4l2rtspserver/libv4l2cpp/src/V4l2MmapDevice.cpp:49
        Device /dev/video2
[NOTICE] /home/username/v4l2rtspserver/libv4l2cpp/src/V4l2MmapDevice.cpp:73
        Device /dev/video2 nb buffer:2
[INFO] /home/username/v4l2rtspserver/libv4l2cpp/src/V4l2MmapDevice.cpp:92
        Device /dev/video2 buffer idx:0 size:2764800 offset:0
[INFO] /home/username/v4l2rtspserver/libv4l2cpp/src/V4l2MmapDevice.cpp:92
        Device /dev/video2 buffer idx:1 size:2764800 offset:0
[NOTICE] /home/username/v4l2rtspserver/src/V4l2RTSPServer.cpp:62
        Create Source .../dev/video2
[NOTICE] /home/username/v4l2rtspserver/inc/BaseServerMediaSubsession.h:49
        format:video/JPEG
[NOTICE] /home/username/v4l2rtspserver/inc/V4l2RTSPServer.h:237
        Play this stream using the URL "rtsp://10.1.1.7:8554/unicast"
[NOTICE] /home/username/v4l2rtspserver/src/V4L2DeviceSource.cpp:96
        begin thread
[DEBUG] /home/username/v4l2rtspserver/src/V4L2DeviceSource.cpp:108
        waitingFrame    delay:1ms

...

[DhandleCmd_SETUP:SETUP rtsp://10.1.1.7:8554/unicast/track1 RTSP/1.0
CSeq: 4
User-Agent: LibVLC/3.0.18 (LIVE555 Streaming Media v2016.11.28)
Transport: RTP/AVP;unicast;client_port=52268-52269

...

[DEBUG] /home/username/v4l2rtspserver/src/V4L2DeviceSource.cpp:143
        Queue is empty
[DEBUG] /home/username/v4l2rtspserver/src/V4L2DeviceSource.cpp:108
        waitingFrame    delay:100ms
[DEBUG] /home/username/v4l2rtspserver/src/V4L2DeviceSource.cpp:224
        postFrame       timestamp:1687302032.125536     size:63228      diff:0ms
[DEBUG] /home/username/v4l2rtspserver/src/V4L2DeviceSource.cpp:257
        queueFrame      timestamp:1687302032.125536     size:63228      diff:0ms
[DEBUG] /home/username/v4l2rtspserver/src/V4L2DeviceSource.cpp:165
        deliverFrame    timestamp:1687302032.128871     size:63228      diff:3ms        queue:0
[DEBUG] /home/username/v4l2rtspserver/src/MJPEGVideoSource.cpp:49
        DQT length:67
[DEBUG] /home/username/v4l2rtspserver/src/MJPEGVideoSource.cpp:57
        Quantization table idx:0 precision:0 size:64 total size:128
[DEBUG] /home/username/v4l2rtspserver/src/MJPEGVideoSource.cpp:49
        DQT length:67
[DEBUG] /home/username/v4l2rtspserver/src/MJPEGVideoSource.cpp:57
        Quantization table idx:1 precision:0 size:64 total size:128
[DEBUG] /home/username/v4l2rtspserver/src/MJPEGVideoSource.cpp:26
        SOF length:17
[INFO] /home/username/v4l2rtspserver/src/MJPEGVideoSource.cpp:42
        width:960 height:720 type:0 precision:8
[DEBUG] /home/username/v4l2rtspserver/src/MJPEGVideoSource.cpp:77
        SOS length:12
[DEBUG] /home/username/v4l2rtspserver/src/MJPEGVideoSource.cpp:87
        headerSize:208
[DEBUG] /home/username/v4l2rtspserver/src/V4L2DeviceSource.cpp:143
        Queue is empty
[DEBUG] /home/username/v4l2rtspserver/src/V4L2DeviceSource.cpp:108
        waitingFrame    delay:100ms
[DEBUG] /home/username/v4l2rtspserver/src/V4L2DeviceSource.cpp:224
        postFrame       timestamp:1687302032.225471     size:63291      diff:0ms
[DEBUG] /home/username/v4l2rtspserver/src/V4L2DeviceSource.cpp:257
        queueFrame      timestamp:1687302032.225471     size:63291      diff:0ms
[DEBUG] /home/username/v4l2rtspserver/src/V4L2DeviceSource.cpp:165
        deliverFrame    timestamp:1687302032.231454     size:63291      diff:5ms        queue:0
[DEBUG] /home/username/v4l2rtspserver/src/MJPEGVideoSource.cpp:49
        DQT length:67
[DEBUG] /home/username/v4l2rtspserver/src/MJPEGVideoSource.cpp:57
        Quantization table idx:0 precision:0 size:64 tVIDIOC_DQBUF: Invalid argument
```

**An important note**: running the **ffmpeg** command doesn't always produce this error. It seems that it is a timing issue. Sometimes I have to run it a few times before it triggers the error, sometimes just once. When the error occurs I have to restart **v4l2rtspserver** as new client connections report timeouts. New client connections are reported in the **v4l2rtspserver** log but the clients report timeouts and no video streams are displayed.

Please advise and let me know if you need any more debug information.

dragulesq, 2023-08-31:
@mpromonet any thoughts on this? any ideas on what I could try differently? Thanks!

mpromonet, 2023-11-05:
Hi,

the error ```VIDIOC_DQBUF: Invalid argument``` seems to indicate the v4l2 device doesnot give more frame. You may look to v4l2loopback, may be upgrading it.

Best Regards,
Michel.