mpromonet/v4l2rtspserver #322: Green ffplay window instead of the camera's feed

hilliao, 2023-12-23:
**Describe the bug**
Using ffplay to show the video on the webcam, the video feed is green.

**To Reproduce**
Steps to reproduce the behavior:
Executing the command `ffplay rtsp://192.168.1.4:8554/unicast`  to start the RTSP server on [the webcam](https://www.amazon.com/gp/product/B09MF22RFW/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)
```
./v4l2rtspserver
log level:500

[NOTICE] /home/hil/git/v4l2rtspserver/main.cpp:269
	Version: 0.3.8 live555 version:2023.11.30
[NOTICE] /home/hil/git/v4l2rtspserver/src/V4l2RTSPServer.cpp:37
	Create V4L2 Source.../dev/video0
[NOTICE] /home/hil/git/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:133
	driver:uvcvideo capabilities:84a00001 mandatory:4000001
[NOTICE] /home/hil/git/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:136
	/dev/video0 support capture
[NOTICE] /home/hil/git/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:139
	/dev/video0 support streaming
[ERROR] /home/hil/git/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:212
	/dev/video0: Cannot set pixelformat to:H264 format is:MJPG
[NOTICE] /home/hil/git/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:225
	/dev/video0:MJPG size:1920x1080 bufferSize:4177920
[NOTICE] /home/hil/git/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:246
	fps:1/30
[NOTICE] /home/hil/git/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:247
	nbBuffer:0
[NOTICE] /home/hil/git/v4l2rtspserver/libv4l2cpp/src/V4l2MmapDevice.cpp:49
	Device /dev/video0
[NOTICE] /home/hil/git/v4l2rtspserver/libv4l2cpp/src/V4l2MmapDevice.cpp:73
	Device /dev/video0 nb buffer:10
[NOTICE] /home/hil/git/v4l2rtspserver/src/V4l2RTSPServer.cpp:62
	Create Source .../dev/video0
[NOTICE] /home/hil/git/v4l2rtspserver/inc/BaseServerMediaSubsession.h:49
	format:video/JPEG
[NOTICE] /home/hil/git/v4l2rtspserver/inc/V4l2RTSPServer.h:240
	Play this stream using the URL "rtsp://192.168.1.4:8554/unicast"
[NOTICE] /home/hil/git/v4l2rtspserver/src/V4L2DeviceSource.cpp:96
handleCmd_SETUP:SETUP rtsp://192.168.1.4:8554/unicast/track1 RTSP/1.0
Transport: RTP/AVP/UDP;unicast;client_port=12660-12661
CSeq: 3
User-Agent: Lavf60.3.100

^C	begin threadSIGINT

[NOTICE] /home/hil/git/v4l2rtspserver/main.cpp:357
```
observe the video feed window:
[image]

**Expected behavior**
Normal video feed shows on Logitech, Inc. Logitech Webcam C925e on a different PC using the same method to build and execute.

**Desktop (please complete the following information):**
 - Ubuntu 23.10 running on AMD Ryzen 7700x.

mpromonet, 2023-12-30:
Hi,

Maybe increasing log level will help you to get more information.

Best Regards,
Michel.

hilliao, 2023-12-30:
I think the webcam is not supported. I tried https://kaustavmukherjee-66179.medium.com/convert-any-webcam-into-ipcam-in-2-minutes-using-rtsp-server-and-ffmpeg-a27414f08af5 and had similar failures.

closed 2023-12-30 by hilliao

mpromonet, 2023-12-31:
Hi,

This may depends on q field and quantification tables (see https://www.rfc-editor.org/rfc/rfc2435#appendix-A). q field is not decoded, code suppose quantification table are sent.

Best Regards,
Michel.