mpromonet/v4l2rtspserver #327: segfault crash on mipsel

gtxaspec, 2024-02-27:
**Describe the bug**
segfault when connecting to stream

**To Reproduce**
compile and rnu

**Expected behavior**
not crashing
**Screenshots**
n/a
**Desktop (please complete the following information):**
master, on mipsel

**Additional context**

```
[root@thingino nfs_root]# ./v4l2rtspserver -s /dev/video1 
log level:500

[NOTICE] /home/aag/repository/v4l2rtspserver/main.cpp:247
	Version: 0.3.8-8-g5cf22b4-dirty live555 version:2024.02.23
[NOTICE] /home/aag/repository/v4l2rtspserver/src/V4l2RTSPServer.cpp:37
	Create V4L2 Source.../dev/video1
[NOTICE] /home/aag/repository/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:133
	driver:v4l2 loopback capabilities:85008003 mandatory:4000001
[NOTICE] /home/aag/repository/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:135
	/dev/video1 support output
[NOTICE] /home/aag/repository/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:136
	/dev/video1 support capture
[NOTICE] /home/aag/repository/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:138
	/dev/video1 support read/write
[NOTICE] /home/aag/repository/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:139
	/dev/video1 support streaming
[ERROR] /home/aag/repository/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:207
	/dev/video1: Cannot set format:HEVC Invalid argument
[NOTICE] /home/aag/repository/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:225
	/dev/video1:H264 size:1920x1080 bufferSize:8294400
[NOTICE] /home/aag/repository/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:246
	fps:1/25
[NOTICE] /home/aag/repository/v4l2rtspserver/libv4l2cpp/src/V4l2Device.cpp:247
	nbBuffer:2
[NOTICE] /home/aag/repository/v4l2rtspserver/libv4l2cpp/src/V4l2MmapDevice.cpp:49
	Device /dev/video1
[NOTICE] /home/aag/repository/v4l2rtspserver/libv4l2cpp/src/V4l2MmapDevice.cpp:73
	Device /dev/video1 nb buffer:2
[NOTICE] /home/aag/repository/v4l2rtspserver/src/V4l2RTSPServer.cpp:62
	Create Source .../dev/video1
[NOTICE] /home/aag/repository/v4l2rtspserver/inc/BaseServerMediaSubsession.h:49
	format:video/H264
[NOTICE] /home/aag/repository/v4l2rtspserver/inc/V4l2RTSPServer.h:275
Segmentation fault
```

strace:
https://pastebin.pl/view/0c38780c

compiled using musl 1.2.4, gcc13

gtxaspec, 2024-03-06:
solved with this patch for future reference:
https://github.com/radredgreen/wyrecam/blob/main/general/package/live555-wyrecam/20-hacky-sdp-snprintf-fix.patch

closed 2024-03-06 by gtxaspec

mpromonet, 2024-03-07:
Hi,

it maybe a same problem fixed in openwrt with https://github.com/openwrt/packages/blob/master/multimedia/v4l2rtspserver/patches/001_live555_sprintf_format ?

Best Regards,
Michel.

gtxaspec, 2024-03-07:
thank you @mpromonet !  cheers!