mpromonet/v4l2rtspserver #335: Latest Docker image won't install on RPI zero (original v1 hardware).

jshep321, 2024-04-29:
**Describe the bug**
Docker image won't install on RPI zero (original v1 hardware).  

**To Reproduce**
Steps to reproduce the behavior:
1. installed docker on raspberry pi OS (aka raspbian)
2. docker pull mpromonet/v4l2rtspserver:latest
3. or try with docker pull --platform linux/arm/v6 mpromonet/v4l2rtspserver:latest

**Expected behavior**
image should pull since there is a linux/arm/v6 version.  In fact there are two, which might be the problem?

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 RPI Zero v1, bullseye
uname -a:  Linux raspberrypi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux

mpromonet, 2024-05-03:
Hi,

It seems arm/v7 & arm/v6 are mixed.
Did you try to pull mpromonet/v4l2rtspserver:latest-arm32v6 ?

Best Regards,
Michel.

mpromonet, 2024-05-06:
Hi,

I made a test on a Raspberry B, it is working since https://github.com/mpromonet/v4l2rtspserver/commit/ef86de39227e0c97ea27de4955cdc6589355e266.
There is one image arm/v6 and one image arm/v7.

Best Regards,
Michel.

jshep321, 2024-05-11:
Yep that is the issue -- it pulls the v7.  When I specify the v6 it runs as expected.   Thanks!

closed 2024-05-11 by jshep321

jshep321, 2024-05-11:
FYI for anyone looking here is a working docker compose file:

services:
  rtspstreamer:
    container_name: rtspstreamer
    privileged: false
    restart: unless-stopped
    image: mpromonet/v4l2rtspserver:latest-arm32v6
    command: -F3 -v #command line parameters
    devices:
      - /dev/video0:/dev/video0
    ports:
      - "8554:8554" #RTSP