mpromonet/v4l2rtspserver #337: Add ONVIF discovery feature?

tuananh-lab, 2024-05-10:
My project is to build an RTSP server for cameras (using rasberry Pi). My server has two main functions: transmitting RTSP to the client and discovering ONVIF devices in the LAN. I don't know if this project is right. Is it a good choice and can I add a discovery feature?

mpromonet, 2024-05-14:
Hi,

You might look to https://github.com/mpromonet/v4l2onvif

Best Regards 
Michel

tuananh-lab, 2024-05-15:
Thank you for your feedback. I want to install and run this server on Rasberry Pi 4 with camera module V1.3. I just need the server to be able to get the camera stream from the Pi. I see in the documentation you need Download UV4L for Pi. I'm having trouble downloading UV4L driver for Pi. Can I use V4L2 driver instead?

mpromonet, 2024-05-18:
Hi @tuananh-lab 

The approach of this project is to use v4L2 interface, driver should not matter till it implement the interface.

Best Regards,
Michel.

tuananh-lab, 2024-05-21:
> Hi @tuananh-lab
> 
> The approach of this project is to use v4L2 interface, driver should not matter till it implement the interface.
> 
> Best Regards, Michel.

my rasberry pi does not directly support h264,MJPEG format so v4l2rtspserver cannot run on Pi. My Pi has libcamera-vid available and it can record videos in .h264 format. I can use libcamera-vid to help the server run on Pi? If there is no other way to run the server on Pi?

tuananh-lab, 2024-06-02:
hello. I'm really having trouble because the rasberry pi using v4l2 only supports the YUV video format but does not support the formats the server is using such as :h264,MJPEG. What do I need to do so that v4l2rtspserver can run on it? My Pi and can stream rtsp stream from pi cam.

tuananh-lab, 2024-06-17:
Hi @mpromonet 
Do you have any solution for my problem? Can you give advice on how to run v4l2onvif and v4l2rtspserver on rasberry pi 4 model B+ and raspi V3 camera module?

mpromonet, 2025-05-24:
Hi,
I added to [v4l2tools](https://github.com/mpromonet/v4l2tools) a new tool https://github.com/mpromonet/v4l2tools/blob/master/src/v4l2libcamera.cpp that read camera using libecamera and write to v4l2 device (let's say v4l2loopback).
Then this v4l2loopback can be used to as H264 input stream of v4l2rtspserver.

Best Regards,
Michel.

closed 2025-05-24 by mpromonet