mpromonet/v4l2rtspserver #340: Camera Microsoft Corp. LifeCam HD-3000 capability problem

wnpllrzodiac, 2024-07-14:
**Describe the bug**
```
./v4l2rtspserver /dev/video0
log level:500

[NOTICE] (main.cpp:247)                 Version: 0.3.10-4-ge9322f1 live555 version:2024.06.26
[NOTICE] (V4l2RTSPServer.cpp:37)        Create V4L2 Source.../dev/video0
[ERROR]  (V4l2Device.cpp:145)           Mandatory capability not available for device:/dev/video0
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor
[ERROR]  (V4l2Device.cpp:83)            Cannot init device:/dev/video0root@panther-x2:~/work/git/v4l2rtspserver/build__#
```

**To Reproduce**
Steps to reproduce the behavior:
./v4l2rtspserver /dev/video0

**Expected behavior**
server running

**device**
aarch64 panther-x2
kernel 6.1.63
ubuntu22.04

**Additional context**
Add any other context about the problem here.

mpromonet, 2024-07-14:
Hi,
It seems /dev/video0 does not exist.
Best Regards 
Michel

wnpllrzodiac, 2024-07-14:
Nop, I can find that ls /dev/video0 exists.
How could I get more infomation to debug?

mpromonet, 2024-07-14:
Hi,
v4l2-ctl may helps you
Best Regards,
Michel

wnpllrzodiac, 2024-07-20:
@mpromonet 

```
v4l2-ctl --list-devices
rockchip-rga (platform:rga):
        /dev/video0

Microsoft® LifeCam HD-3000: Mi (usb-xhci-hcd.0.auto-1):
        /dev/video1
        /dev/video2
        /dev/media0
```

wnpllrzodiac, 2024-07-20:
`./v4l2rtspserver /dev/video1`

problem soved.

closed 2024-07-20 by wnpllrzodiac