mpromonet/v4l2rtspserver #341: TLS support in V4l2rtspserver

manivalaguru, 2024-08-08:
Hi,
I'm working on enabling streaming with TLS, but I'm encountering issues with the proper initialization.

OS:ubuntu 22.04
server pipeline:
sudo ./v4l2rtspserver /dev/video0 -fMJPG -W1920 -H1080 -P443 -F60 -u routecam -x server.includesprivatekey.pem

client pipeline:
 gst-launch-1.0 rtspsrc location=rtsps://192.168.1.126:443/routecam tls-validation-flags=generic-error protocols=tls latency=0 ! rtpjpegdepay ! jpegdec ! videoconvert ! fpsdisplaysink

While debugging, we identified that the issue occurs because the fTLS.tlsAcceptIsNeeded flag is being set to True even after it has been set to False following the SSL acceptance in the call acceptResult = fInputTLS->accept(fOurSocket);.

`GenericMediaServer::ClientConnection
::ClientConnection(GenericMediaServer& ourServer,
		   int clientSocket, struct sockaddr_storage const& clientAddr,
		   Boolean useTLS)
  : fOurServer(ourServer), fOurSocket(clientSocket), fClientAddr(clientAddr), fTLS(envir()) {
  fInputTLS = fOutputTLS = &fTLS;

  // Add ourself to our 'client connections' table:
  fOurServer.fClientConnections->Add((char const*)this, this);
   fInputTLS->tlsAcceptIsNeeded = True; 
  if (useTLS) {
printf("coming to TLS\n");
    // Perform extra processing to handle a TLS connection:
    fTLS.setCertificateAndPrivateKeyFileNames(ourServer.fTLSCertificateFileName,
					      ourServer.fTLSPrivateKeyFileName);
    fTLS.isNeeded = True;

    fTLS.tlsAcceptIsNeeded = True; // call fTLS.accept() the next time the socket is readable
  }
`
On the client side, this issue keeps recurring.
Progress: (open) Retrieving media info
Progress: (connect) Connecting to rtsps://192.168.1.126:443/routecam
[tls_failre.zip](https://github.com/user-attachments/files/16547956/tls_failre.zip)

I have attached the client-side log and the Wireshark log to help you better understand the issue.
[tls_failre.zip](https://github.com/user-attachments/files/16547957/tls_failre.zip)

Regards,
Manikandan

mpromonet, 2024-08-10:
Hi @manivalaguru 

Did the file contain private keys and certificate ?

Best Regards 
Michel

manivalaguru, 2024-08-12:
Yes,The attached file, server.includesprivatekey.pem, is a valid TLS certificate containing both the private key and certificate. Could you provide a step-by-step guide on how to test TLS connectivity using this certificate? 
[key_cert.zip](https://github.com/user-attachments/files/16580125/key_cert.zip)

manivalaguru, 2024-08-13:
Any inputs?

mpromonet, 2024-09-28:
Hi @manivalaguru 
I proposed a modification in VLC https://code.videolan.org/videolan/vlc/-/merge_requests/6050 to allow to connect to RTSPS.
Best Regards,
Michel.

manivalaguru, 2025-03-10:
Hi i have tired in latest VLC and still facing the issue i need to stream in RTSPS

server pipeline:
 ./v4l2rtspserver /dev/video0 -fMJPG -W1280 -H720 -F60 -u routecam -P443 -X -x keycert.pem

on VLC
vlc -vvv rtsps://192.168.1.126:443/routecam --file-logging --logfile=vlc-log.txt

I have attached my vlc error log kindly check them

[error_logs.zip](https://github.com/user-attachments/files/19164722/error_logs.zip)

manivalaguru, 2025-03-11:
Any inputs?