mpromonet/v4l2rtspserver #345: Can't get RTSP stream using RPi node's hostname, but works with its IP address

neerajkc, 2024-10-16:
I have a RPi camera connected to RPi zero 2w on LAN and v4l2rtspserver setup. I am able to receive the video stream on "rtsp://192.168.1.44:8554/unicast", but can't get a connection on "rtsp://rpiz2w1:8554/unicast". I also tried "rtsp://rpiz2w1.local:8554/unicast" but that also doesn't work. Here, IP address is DHCP assigned by router and hostname is rpiz2w1. I am able to ping or ssh into the RPi using both ip address or hostname from my PC, but RTSP is not going through. Any idea how to resolve this? My usecase requires connecting to several RPi+camera nodes via their hostnames since IP addresses may change (say after reboots).

mpromonet, 2024-10-16:
Hi @neerajkc 
You need to configure somehow the name resolution, using host file, DNS, mDNS, or whatever.... for instance you may like to avahi.
Best Regards,
Michel.

neerajkc, 2024-10-17:
Indeed, I successfully get the stream by hostname on Windows, iPhone, and even on another RPi 4b on the LAN. The problem seems to be on my Ubuntu 22.04 PC. I tried to play around with Avahi according to this [wiki](https://wiki.archlinux.org/title/Avahi). It basically points to the clashes between systemd-resolved (enabled by default) and avahi-daemon services when it comes to mDNS. Tried suggestions on this page, nothing works. Same problem reported on many forums.
The ran the following command on my system: 
`host -t SOA local`
and it returned:
`Host local not found: 2(SERVFAIL)`.
Whereas on RPi 4b it returns correctly:
`Host local not found: 3(NXDOMAIN)`