mpromonet/v4l2rtspserver #346: Cannot init device:/dev/video1

lyd107, 2024-11-20:
**Describe the bug**
A clear and concise description of what the bug is.

I have met this error when I run the shell code, I successfully compiled.

```shell
neardi@LPA3588:~/v4l2rtspserver$ ./v4l2rtspserver -H 480 -W 640 -F 30 -P 8554 /dev/video1
log level:500

[NOTICE] (main.cpp:249)                	Version: 0.3.11-4-g410c0d4-dirty live555 version:2024.08.01
[NOTICE] (V4l2RTSPServer.cpp:37)       	Create V4L2 Source.../dev/video1--driver--:rkcif

[ERROR]  (V4l2Device.cpp:146)          	Mandatory capability not available for device:/dev/video1
VIDIOC_STREAMOFF: Bad file descriptor
VIDIOC_REQBUFS: Bad file descriptor
[ERROR]  (V4l2Device.cpp:83)           	Cannot init device:/dev/video1
```

camera info : AHD camera
pc info         : RK3588

dmacattack, 2024-11-20:
whats the output of :
``` bash
 v4l2-ctl --device=/dev/video1 --all
```
and :

``` bash
 v4l2-ctl --device=/dev/video0 --all
```

commonly v4l2 drivers have multiple associated nodes, with different capabilities.

lyd107, 2024-11-22:
> whats the output of :
> 
> ```shell
>  v4l2-ctl --device=/dev/video1 --all
> ```
> 
> and :
> 
> ```shell
>  v4l2-ctl --device=/dev/video0 --all
> ```
> 
> commonly v4l2 drivers have multiple associated nodes, with different capabilities. You could be using the wrong one

> ```shell
> v4l2-ctl --device=/dev/video0 --all
> ```
the output:
Driver Info:
	Driver name      : rkcif
	Card type        : rkcif
	Bus info         : platform:rkcif-mipi-lvds
	Driver version   : 5.10.110
	Capabilities     : 0x84201000
		Video Capture Multiplanar
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04201000
		Video Capture Multiplanar
		Streaming
		Extended Pix Format
Media Driver Info:
	Driver name      : rkcif
	Model            : rkcif-mipi-lvds
	Serial           : 
	Bus info         : 
	Media version    : 5.10.110
	Hardware revision: 0x00000000 (0)
	Driver version   : 5.10.110
Interface Info:
	ID               : 0x03000002
	Type             : V4L Video
Entity Info:
	ID               : 0x00000001 (1)
	Name             : stream_cif_mipi_id0
	Function         : V4L2 I/O
	Pad 0x01000004   : 0: Sink
	  Link 0x02000046: from remote pad 0x100002f of entity 'rockchip-mipi-csi2': Data, Enabled
	  Link 0x0200005c: from remote pad 0x1000030 of entity 'rockchip-mipi-csi2': Data
	  Link 0x02000072: from remote pad 0x1000031 of entity 'rockchip-mipi-csi2': Data
	  Link 0x02000088: from remote pad 0x1000032 of entity 'rockchip-mipi-csi2': Data
	  Link 0x0200009e: from remote pad 0x1000033 of entity 'rockchip-mipi-csi2': Data
	  Link 0x020000b4: from remote pad 0x1000034 of entity 'rockchip-mipi-csi2': Data
	  Link 0x020000ca: from remote pad 0x1000035 of entity 'rockchip-mipi-csi2': Data
	  Link 0x020000e0: from remote pad 0x1000036 of entity 'rockchip-mipi-csi2': Data
	  Link 0x020000f6: from remote pad 0x1000037 of entity 'rockchip-mipi-csi2': Data
	  Link 0x0200010c: from remote pad 0x1000038 of entity 'rockchip-mipi-csi2': Data
	  Link 0x02000122: from remote pad 0x1000039 of entity 'rockchip-mipi-csi2': Data
Priority: 2
Format Video Capture Multiplanar:
	Width/Height      : 1920/1080
	Pixel Format      : 'RG10' (10-bit Bayer RGRG/GBGB)
	Field             : None
	Number of planes  : 1
	Flags             : 
	Colorspace        : Default
	Transfer Function : Default
	YCbCr/HSV Encoding: Default
	Quantization      : Default
	Plane 0           :
	   Bytes per Line : 2560
	   Size Image     : 2764800
Selection Video Capture: crop, Left 0, Top 0, Width 1920, Height 1080, Flags: 
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 1920, Height 1080, Flags: 
Selection Video Output: crop, Left 0, Top 0, Width 1920, Height 1080, Flags: 
Selection Video Output: crop_bounds, Left 0, Top 0, Width 1920, Height 1080, Flags:

> ```shell
> v4l2-ctl --device=/dev/video1 --all
> ```
the output:
Driver Info:
	Driver name      : rkcif
	Card type        : rkcif
	Bus info         : platform:rkcif-mipi-lvds
	Driver version   : 5.10.110
	Capabilities     : 0x84201000
		Video Capture Multiplanar
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04201000
		Video Capture Multiplanar
		Streaming
		Extended Pix Format
Media Driver Info:
	Driver name      : rkcif
	Model            : rkcif-mipi-lvds
	Serial           : 
	Bus info         : 
	Media version    : 5.10.110
	Hardware revision: 0x00000000 (0)
	Driver version   : 5.10.110
Interface Info:
	ID               : 0x03000006
	Type             : V4L Video
Entity Info:
	ID               : 0x00000005 (5)
	Name             : stream_cif_mipi_id1
	Function         : V4L2 I/O
	Pad 0x01000008   : 0: Sink
	  Link 0x02000048: from remote pad 0x100002f of entity 'rockchip-mipi-csi2': Data
	  Link 0x0200005e: from remote pad 0x1000030 of entity 'rockchip-mipi-csi2': Data, Enabled
	  Link 0x02000074: from remote pad 0x1000031 of entity 'rockchip-mipi-csi2': Data
	  Link 0x0200008a: from remote pad 0x1000032 of entity 'rockchip-mipi-csi2': Data
	  Link 0x020000a0: from remote pad 0x1000033 of entity 'rockchip-mipi-csi2': Data
	  Link 0x020000b6: from remote pad 0x1000034 of entity 'rockchip-mipi-csi2': Data
	  Link 0x020000cc: from remote pad 0x1000035 of entity 'rockchip-mipi-csi2': Data
	  Link 0x020000e2: from remote pad 0x1000036 of entity 'rockchip-mipi-csi2': Data
	  Link 0x020000f8: from remote pad 0x1000037 of entity 'rockchip-mipi-csi2': Data
	  Link 0x0200010e: from remote pad 0x1000038 of entity 'rockchip-mipi-csi2': Data
	  Link 0x02000124: from remote pad 0x1000039 of entity 'rockchip-mipi-csi2': Data
Priority: 2
Format Video Capture Multiplanar:
	Width/Height      : 1920/1080
	Pixel Format      : 'RG10' (10-bit Bayer RGRG/GBGB)
	Field             : None
	Number of planes  : 1
	Flags             : 
	Colorspace        : Default
	Transfer Function : Default
	YCbCr/HSV Encoding: Default
	Quantization      : Default
	Plane 0           :
	   Bytes per Line : 2560
	   Size Image     : 2764800
Selection Video Capture: crop, Left 0, Top 0, Width 1920, Height 1080, Flags: 
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 1920, Height 1080, Flags: 
Selection Video Output: crop, Left 0, Top 0, Width 1920, Height 1080, Flags: 
Selection Video Output: crop_bounds, Left 0, Top 0, Width 1920, Height 1080, Flags:

dmacattack, 2024-12-16:
it looks like maybe your rockchip isp isnt setup. Its been a while since ive done it on rockchip. 

https://www.kernel.org/doc/html/v5.17/admin-guide/media/rkisp1.html

fundamentally, after your ISP is setup, your v4l2 device should be usable without v4l2rtspserver 
``` bash
$ gst-launch-1.0 v4l2src --device=/dev/video0 ! videoconvert ! autovideosink
```
this should give _some_ type of video stream. once that works, the rtsp server should work too