mpromonet/v4l2rtspserver #348: Frame missing in h264 &h265 streaming

manivalaguru, 2025-02-17:
Hi,
While streaming in H.265 (HEVC) format, we intermittently encounter grey frames. Upon enabling debugging in VLC, we observe the following error when the issue occurs:

"Could not find ref with POC 17."

May we know 
1) Why is this happening
2)How to resolve them

Regards
Manikandan

manivalaguru, 2025-02-17:
Also there Is there any way to increase  the encoding buffer size to Maximum per encoding channel

manivalaguru, 2025-02-20:
Any inputs

mpromonet, 2025-02-21:
Hi @manivalaguru 

You may increase the log level, try to read your device with vlc or ffmpeg to see if it's behave the same.
It might be missing in VPS, SPS, pps processing.

Best Regards
Michel

manivalaguru, 2025-02-24:
Hi,
This is the vlc log

[000079855004acd0] avcodec decoder warning: More than 11 late frames, dropping frame
[hevc @ 0x79854402c300] nal_unit_type: 1(TRAIL_R), nuh_layer_id: 0, temporal_id: 0
[hevc @ 0x79854402c300] Could not find ref with POC 4
[hevc @ 0x79854402c300] Output frame with POC 5.
[hevc @ 0x79854403c8c0] Decoded frame with POC 26.
[000079855004acd0] avcodec decoder warning: More than 11 late frames, dropping frame
[hevc @ 0x79854403c8c0] nal_unit_type: 1(TRAIL_R), nuh_layer_id: 0, temporal_id: 0
[hevc @ 0x79854403c8c0] Could not find ref with POC 6
[hevc @ 0x79854403c8c0] Output frame with POC 7.
[hevc @ 0x79854404cf40] Decoded frame with POC 27.
[000079855004acd0] avcodec decoder warning: More than 11 late frames, dropping frame
[hevc @ 0x79854404cf40] nal_unit_type: 1(TRAIL_R), nuh_layer_id: 0, temporal_id: 0
[hevc @ 0x79854404cf40] Could not find ref with POC 8
[hevc @ 0x79854404cf40] Output frame with POC 9.
[hevc @ 0x79854407e880] Decoded frame with POC 0.
[hevc @ 0x79854405d640] Decoded frame with POC 28.
[000079855004acd0] avcodec decoder warning: More than 11 late frames, dropping frame
[hevc @ 0x79854405d640] nal_unit_type: 1(TRAIL_R), nuh_layer_id: 0, temporal_id: 0
[hevc @ 0x79854405d640] Could not find ref with POC 10
[hevc @ 0x79854405d640] Output frame with POC 11.
[hevc @ 0x79854408f100] Decoded frame with POC 1.
[hevc @ 0x79854403c8c0] Decoded frame with POC 7.
[hevc @ 0x79854406df40] Decoded frame with POC 29.
[000079855004acd0] avcodec decoder warning: More than 11 late frames, dropping frame
[000079855004acd0] avcodec decoder warning: More than 11 late frames, dropping frame
[hevc @ 0x79854406df40] nal_unit_type: 1(TRAIL_R), nuh_layer_id: 0, temporal_id: 0
[hevc @ 0x79854406df40] Could not find ref with POC 12
[hevc @ 0x79854406df40] Output frame with POC 13.
[000079855004acd0] avcodec decoder warning: More than 11 late frames, dropping frame
[hevc @ 0x79854407e880] nal_unit_type: 1(TRAIL_R), nuh_layer_id: 0, temporal_id: 0
[hevc @ 0x79854407e880] Could not find ref with POC 14
[hevc @ 0x79854407e880] Output frame with POC 15.
[hevc @ 0x79854402c300] Decoded frame with POC 5.
[hevc @ 0x79854408f100] nal_unit_type: 1(TRAIL_R), nuh_layer_id: 0, temporal_id: 0
[hevc @ 0x79854408f100] Could not find ref with POC 16
[hevc @ 0x79854408f100] Output frame with POC 17.
[hevc @ 0x79854404cf40] Decoded frame with POC 9.
[hevc @ 0x79854409fa40] Decoded frame with POC 2.
[000079855004acd0] avcodec decoder warning: More than 11 late frames, dropping frame
[hevc @ 0x79854409fa40] nal_unit_type: 1(TRAIL_R), nuh_layer_id: 0, temporal_id: 0
[hevc @ 0x79854409fa40] Could not find ref with POC 18
[hevc @ 0x79854409fa40] Output frame with POC 19.
[hevc @ 0x79854406df40] Decoded frame with POC 13.
[hevc @ 0x79854400b8c0] Decoded frame with POC 3.
[000079855004acd0] avcodec decoder warning: More than 11 late frames, dropping frame
[hevc @ 0x79854400b8c0] nal_unit_type: 1(TRAIL_R), nuh_layer_id: 0, temporal_id: 0
[000079855004acd0] avcodec decoder warning: More than 11 late frames, dropping frame
[hevc @ 0x79854400b8c0] Could not find ref with POC 20
[hevc @ 0x79854400b8c0] Output frame with POC 21.
[000079855004acd0] avcodec decoder warning: More than 11 late frames, dropping frame
[hevc @ 0x79854407e880] Decoded frame with POC 15.
[hevc @ 0x79854402c300] nal_unit_type: 1(TRAIL_R), nuh_layer_id: 0, temporal_id: 0
[hevc @ 0x79854402c300] Could not find ref with POC 22
[hevc @ 0x79854402c300] Output frame with POC 23.
[hevc @ 0x79854403c8c0] nal_unit_type: 1(TRAIL_R), nuh_layer_id: 0, temporal_id: 0
[hevc @ 0x79854403c8c0] Could not find ref with POC 24

we have verifed at the encoder side we seeing no such frame drops in vlc or ffmpeg streaming only we are seeing this. Any idea why may this happen?

mpromonet, 2025-02-26:
Hi @manivalaguru 

Maybe i-frames are big and are dropped by your network. You can discriminate this using rtp over tcp.

Best regards
Michel