mpromonet/v4l2rtspserver #349: Issue with Multicast Streaming in v4l2rtspserver

manivalaguru, 2025-02-24:
Hi,
We are attempting to enable multicast streaming using v4l2rtspserver.
Sender Configuration:

Command executed on the device:

/tmp/v4l2rtspserver -m multicast -M 239.255.42.42:5004 /dev/video3 -fHEVC

Log Output:

[NOTICE] (main.cpp:254)                 Version: 0.3.11-21-gb0d24eb-dirty live555 version:2024.10.31
VIDIOC_REQBUFS: Inappropriate ioctl for device
VIDIOC_STREAMOFF: Inappropriate ioctl for device
VIDIOC_REQBUFS: Inappropriate ioctl for device
[NOTICE] (V4l2RTSPServer.cpp:37)        Create V4L2 Source...multicast
[NOTICE] (V4l2RTSPServer.h:142)         RTP  address 232.221.213.121:20000
[NOTICE] (V4l2RTSPServer.h:143)         RTCP address 232.221.213.121:20001
...
[NOTICE] (V4l2RTSPServer.h:275)         Play this stream using the URL "rtsp://192.168.1.10:8554/video3_multicast"

Receiver Configuration:

Command executed on the receiving side:

vlc -vvv rtsp://192.168.1.10:8554/video3_multicast

Log Output:

[hevc @ 0x73ebb0074340] Decoding PPS
[000073ebb004cc70] avcodec decoder debug: codec (hevc) started
[000073ebb004cc70] avcodec decoder debug: using frame thread mode with 9 threads
[000073ebb004cc70] main decoder debug: using video decoder module "avcodec"
...
[000073ebb4000c40] main input debug: `rtsp://192.168.1.10:8554/video3_multicast' successfully opened

Issue:

    The stream opens successfully in VLC, but only a black screen is displayed.
    Unicast streaming works without any issues.

Troubleshooting Attempts:

    Verified the video device is functional.
    Ensured multicast routing is enabled and IGMP snooping is properly configured.
    Checked firewall settings to allow multicast traffic.
    Confirmed network supports multicast.

manivalaguru, 2025-02-24:
Also is there way to know if the multicast packet is being sent on the sender

mpromonet, 2025-03-01:
Hi @manivalaguru 

As you can see in the log the option  `-M 239.255.42.42:5004` is not take into account, we see ` RTP address 232.221.213.121:20000`.
This is because -M has default value, you should remove the sapce and use `-M239.255.42.42:5004`  to specify your multicast group & port. It's the same for `-m`
tshark or tcldump or strace should helps you to know if packet are sent 

Best Regards,
Michel.

manivalaguru, 2025-03-11:
Thanks it worked

closed 2025-03-11 by mpromonet