mpromonet/v4l2rtspserver #350: enabling SRTP or RTSPS in cross compilation.

manivalaguru, 2025-03-11:
Hi,

I am attempting to enable RTSPS in my RISCV-based architecture. However, we cross-compile the v4l2rtspserver for RISCV, and I have noticed that the following options are not appearing:

    -x <sslkeycert> : Enable SRTP
    -X : Enable RTSPS

Could you please confirm if any additional configurations or dependencies need to be enabled before cross-compiling?

Below is the output of the help command for both the x86 and RISCV builds:
x86 Help Command:

`./v4l2rtspserver [-v[v]] [-Q queueSize] [-O file]  
                  [-I interface] [-P RTSP port] [-p RTSP/HTTP port] [-m multicast url] [-u unicast url] [-M multicast addr] [-c] [-t timeout] [-T] [-S[duration]]  
                  [-r] [-w] [-s] [-f[format] [-W width] [-H height] [-F fps] [device] [device]  
 -v               : verbose  
 -vv              : very verbose  
 -Q <length>      : Number of frame queue (default 5)  
 -O <output>      : Copy captured frame to a file or a V4L2 device  
 -b <webroot>     : Path to webroot  
 **RTSP/RTP options:**  
 -I <addr>        : RTSP interface (default autodetect)  
 -P <port>        : RTSP port (default 8554)  
 -p <port>        : RTSP over HTTP port (default 0)  
 -U <user>:<pass> : RTSP user and password  
 -R <realm>       : Use md5 password 'md5(<username>:<realm>:<password>)'  
 -u <url>         : Unicast URL (default unicast)  
 -m <url>         : Multicast URL (default multicast)  
 -M <addr>        : Multicast group:port (default is random_address:20000)  
 -c               : Do not repeat config (default: repeat config before IDR frame)  
 -t <timeout>     : RTCP expiration timeout in seconds (default 65)  
 -S[<duration>]   : Enable HLS & MPEG-DASH with segment duration in seconds (default 2)  
 -x <sslkeycert>  : Enable SRTP  
 -X               : Enable RTSPS  `

RISCV Help Command:

root@routecam:~# ./v4l2rtspserver /dev/video0 -fMJPG -W1280 -H720 -P8554 -F30 -u routecam -X -x keycert.pem -U econ:econ  
`./v4l2rtspserver [-v[v]] [-Q queueSize] [-O file]  
                  [-I interface] [-P RTSP port] [-p RTSP/HTTP port] [-m multicast url] [-u unicast url] [-M multicast addr] [-c] [-t timeout] [-T] [-S[duration]]  
                  [-r] [-w] [-s] [-f[format] [-W width] [-H height] [-F fps] [device] [device]  
 -v               : verbose  
 -vv              : very verbose  
 -Q <length>      : Number of frame queue (default 5)  
 -O <output>      : Copy captured frame to a file or a V4L2 device  
 -b <webroot>     : Path to webroot  
 **RTSP/RTP options:**  
 -I <addr>        : RTSP interface (default autodetect)  
 -P <port>        : RTSP port (default 8554)  
 -p <port>        : RTSP over HTTP port (default 0)  
 -U <user>:<pass> : RTSP user and password  
 -R <realm>       : Use md5 password 'md5(<username>:<realm>:<password>)'  
 -u <url>         : Unicast URL (default routecam)  
 -m <url>         : Multicast URL (default multicast)  
 -M <addr>        : Multicast group:port (default is random_address:20000)  
 -c               : Do not repeat config (default: repeat config before IDR frame)  
 -t <timeout>     : RTCP expiration timeout in seconds (default 65)  
 -S[<duration>]   : Enable HLS & MPEG-DASH with segment duration in seconds (default 2)  `

It appears that the -x and -X options are missing from the RISCV build. Could you advise on whether additional dependencies or compilation flags are required to enable these options?

Thanks in advance for your support.

manivalaguru, 2025-03-12:
Hi ,
Any input?

manivalaguru, 2025-03-12:
Hi,
I was able to resolve the issue by installing the server with OpenSSL enabled.
Server Setup:

./v4l2rtspserver /dev/video3 -fH264 -W1280 -H720 -P 5005 -F30 -u routecam -X -x keycert.pem -U econ:econ

Log Level: 500

Logs:
[NOTICE] (main.cpp:254) Version: 0.3.11-23-gde3c395-dirty live555 version: 2025.01.17
[NOTICE] (V4l2RTSPServer.cpp:37) Creating V4L2 Source... /dev/video3
[NOTICE] (BaseServerMediaSubsession.h:49) RTP format: video/H264

RTSP SETUP Request:
handleCmd_SETUP: SETUP rtsps://192.168.1.10:5005/routecam/track1 RTSP/1.0
Transport: RTP/AVP/TCP;unicast;interleaved=0-1
CSeq: 4
User-Agent: Lavf60.16.100
Authorization: Digest username="econ", realm="LIVE555 Streaming Media", nonce="2e6d1a9e4c5903c4e37c657eebe848c7", uri="rtsps://192.168.1.10:5005/routecam/track1", response="57210005a0689f9c7c9e0d00b2a7fda5"

I have tested the setup using FFmpeg, VLC, and GStreamer and have attached all error logs in error_log.zip for reference.
Software Versions:

    FFmpeg:
    ffmpeg --version
    Version: 6.1-1build2~22.04

    GStreamer:
    gst-inspect-1.0 --version
    Version: 1.20.3

    VLC:
    /snap/bin/vlc --version
    Version: 3.0.20 Vetinari

Could you please review and let me know if I am missing anything?

[error_log.zip](https://github.com/user-attachments/files/19211725/error_log.zip)

manivalaguru, 2025-03-13:
FFMPEG PIPELINE:

ffmpeg -udp -i "rtsps://econ:econ@192.168.1.10:5005/routecam"
ffmpeg -rtsp_transport tcp -probesize 10000000 -analyzeduration 20000000 -i "rtsps://econ:econ@192.168.1.10:5005/routecam"

Gstreamer:
GST_DEBUG=5 gst-launch-1.0 rtspsrc location=rtsps://econ:econ@192.168.1.10:5005/routecam tls-validation-flags=0 protocols=tcp latency=100     ! decodebin ! autovideosink

Vlc:
Vlc -vvv rtsps://econ:econ@192.168.1.10:5005/routecam

manivalaguru, 2025-03-14:
Hi any update

manivalaguru, 2025-03-17:
Hi have you got time to check on this

mpromonet, 2025-03-17:
Hi @manivalaguru 
You should read contributing.md.
Feel free to submit a PR with your fix.
Best regards.
Michel

manivalaguru, 2025-03-17:
hi @mpromonet 

I’m still unable to get the stream working over RTSPS. Could you confirm if the server has been tested with RTSPS?

I'm running the following command on the server:

./v4l2rtspserver /dev/video3 -fH264 -W1280 -H720 -P 5005 -F30 -u routecam -X -x keycert.pem -U econ:econ

However, I’m unable to stream using VLC, GStreamer, or FFmpeg.

Could you please share details on how you have tested RTSPS streaming or suggest any necessary configurations?

mpromonet, 2025-03-17:
Hi @manivalaguru 
I tested it with vlc I compiled enabling openssl. I made PR to vlc that were refused.
Best Regards 
Michel

manivalaguru, 2025-03-17:
Thank you for your response.

Would compiling OpenSSL with VLC version 3.0.20 be sufficient, or is a newer version required? Could you also share which version you have tested?

Additionally, since this server includes MIKEY, I believe we need to provide the generated SRTP key on the client side in the GStreamer pipeline, as shown below. Do you have any insights or experience with this approach?

gst-launch-1.0 rtspsrc location=rtsps://econ:econ@192.168.0.171:5005/routecam tls-validation-flags=0 protocols=udp \
    ! queue \
    ! application/x-srtp, srtp-key=file://srtp_key.bin, srtp-cipher=aes-128-icm, srtp-auth=hmac-sha1-80, roc=0 \
    ! decodebin \
    ! autovideosink -v

Looking forward to your guidance.

manivalaguru, 2025-03-17:
Hi i am using this to save the file in OPENRTSP 
openRTSP -V -b 100000 -q -w 1280 -h 720 -f 30 -u econ econ -p 5004 "rtsps://192.168.0.171:5005/routecam" > new.mp4 but could not save the file can you give us the command you have used

manivalaguru, 2025-03-18:
Hi Any inputs,

Vlc version you have cross compile with openssl to test server?
How can i construct a pipeline with MIKEY for gstreamer?
Pipeline you have used in OPENRTSP?

this is my server pipeline:
 ./v4l2rtspserver /dev/video3 -fMJPG -W1920 -H1080 -P 5005 -F30 -u routecam -X -x server.pem -U econ:econ
Note: i am using self signed certificate

manivalaguru, 2025-03-20:
Hi,
I can able to compile vlc with openssl and live555 but my preview closes after an second can you check my log file and let me know if am missing something

manivalaguru, 2025-03-24:
Hi reminder,
May i know the answer to this questions,
1.)Vlc version you have cross compile with openssl to test server?
2.)How can i construct a pipeline with MIKEY for gstreamer?
 Since this server includes MIKEY, I believe we need to provide the generated SRTP key on the client side in the GStreamer pipeline, as shown below. Do you have any insights or experience with this approach?

gst-launch-1.0 rtspsrc location=rtsps://econ:econ@192.168.0.171:5005/routecam tls-validation-flags=0 protocols=udp
! queue
! application/x-srtp, srtp-key=file://srtp_key.bin, srtp-cipher=aes-128-icm, srtp-auth=hmac-sha1-80, roc=0
! decodebin
! autovideosink -v
i could not get the mickey anywhere on the SDP on the receiving side
3.)Pipeline you have used in OPENRTSP?

this is my server pipeline:
./v4l2rtspserver /dev/video3 -fMJPG -W1920 -H1080 -P 5005 -F30 -u routecam -X -x server.pem -U econ:econ
Note: i am using self signed certificate

i can able to compile vlc after multiple tries with 

[vlc-log.txt](https://github.com/user-attachments/files/19416574/vlc-log.txt)

openssl and live555 but streaming closed after a couple this is log file kindly check that

manivalaguru, 2025-03-24:
Hi @mpromonet 
I wanted to follow up regarding the following questions:
1)which VLC Version you are using for testing?

2)How can I construct a GStreamer pipeline that includes MIKEY? Since the server supports MIKEY, I assume we need to provide the generated SRTP key on the client side within the GStreamer pipeline, as shown below:
gst-launch-1.0 rtspsrc location=rtsps://econ:econ@192.168.0.171:5005/routecam tls-validation-flags=0 protocols=udp \
! queue \
! application/x-srtp, srtp-key=file://srtp_key.bin, srtp-cipher=aes-128-icm, srtp-auth=hmac-sha1-80, roc=0 \
! decodebin \
! autovideosink -v
However, I am unable to find the MIKEY information within the SDP on the receiving side. Do you have any insights or experience with this approach?

3)Could you share the pipeline configuration you have used in OpenRTSP?
this is the pipeline i am using which is not working:
openRTSP -b 100000 -D 1 -P 4 -d 60 -u econ econ -F recording "rtsps://192.168.0.171:5005/routecam"

My server pipeline:

./v4l2rtspserver /dev/video3 -fMJPG -W1920 -H1080 -P 5005 -F30 -u routecam -X -x server.pem -U econ:econ

Note: I am using a self-signed certificate.

Additionally, after multiple attempts, I was able to compile VLC with OpenSSL and Live555, but the stream closes after a short time. I have attached the log file (vlc-log.txt) for your review. Could you please check and provide any recommendations?

Looking forward to your response.

[vlc-log.txt](https://github.com/user-attachments/files/19416691/vlc-log.txt)

mpromonet, 2025-03-29:
Hi @manivalaguru 
It was on 4.x, last time I checked, rtsps is not supported in 3.x (see https://code.videolan.org/videolan/vlc/-/merge_requests/6050)
Best Regards,
Michel.

manivalaguru, 2025-03-31:
Hi,
I have tested it on 3.0.20 which is working fine. Can you help regarding gstreamer?