mpromonet/v4l2rtspserver #351: Walkthrough or raspberry pi and bookworm

d-a-v, 2025-04-27:
**Is your feature request related to a problem? Please describe.**

I am frustrated to not beeing able to install and use v4l2rtspserver on old rPi2 and rPi-camera as easily as before.

Libcamera is now the default (#283 won't be fixed) and I'm struggling to find how to enable old camera driver/api while staying on bookworm (because old OSes are meant to be unsupported at some point). Should I ditch my hardware ?

There a similar question in discussion (#339) without answer.

**Describe the solution you'd like**

Has anyone had success in running v4l2rtspserver on raspberry pi / raspi camera / bookworm(debian12) ?

Should we be limited to bullseye(debian11)  ?

Thanks

**Describe alternatives you've considered**

A lot of tries with poorly documented transition between APIs.
Spicifically I don't hame the camera menu in the raspi-config tool.

mpromonet, 2025-04-27:
Hi @d-a-v 

There is the option of using libcamera-v4l2, but it is limited to video/RAW format.
Another option could be to write https://github.com/raspberrypi/rpicam-apps/blob/main/apps/rpicam_vid.cpp like application that write to a [v4l2loopback ](https://github.com/v4l2loopback/v4l2loopback) device that feeds v4l2rtspserver.

Best Regards,
Michel.

mpromonet, 2025-05-11:
Hi @d-a-v 

I made some experiment in https://github.com/mpromonet/v4l2tools/blob/master/src/v4l2libcamera.cpp that might help you ?

Best Regards,
Michel.

jcfain, 2025-05-16:
> Hi [@d-a-v](https://github.com/d-a-v)
> 
> I made some experiment in https://github.com/mpromonet/v4l2tools/blob/master/src/v4l2libcamera.cpp that might help you ?
> 
> Best Regards, Michel.

I just happened to be trying to use this on the latest raspbian os today. I found this post but im not sure how to use it. I was able to build the tools on my system but I dont think that file was included in the build.

How would I go about testing it?

mpromonet, 2025-05-17:
Hi @jcfain 

It build with make if libcamera-dev is installed. Then it needs v4l2loopback device /dev/videoX and v4l2libcamera -o /dev/videoX will write camera h264 stream that could be used as v4l2rtspstreamer input.

Best Regards 
Michel

jcfain, 2025-05-17:
Ive built and installed it a few times now even cleaning and blowing the whole git dir away. No v4l2libcamera bin. 
In /usr/bin, I only have: 
```
v4l2-compliance
v4l2compress
v4l2copy
v4l2-ctl
v4l2dump
v4l2source_yuv
v4l2-sysfs-path
```
Not sure what I'm doing wrong.

mpromonet, 2025-05-17:
Hi @jcfain 

You should look to https://github.com/mpromonet/v4l2tools/blob/master/Makefile#L53

Best Regards,
Michel.

jcfain, 2025-05-17:
> Hi [@jcfain](https://github.com/jcfain)
> 
> You should look to https://github.com/mpromonet/v4l2tools/blob/master/Makefile#L53
> 
> Best Regards, Michel.

Ya I was editing my last post and you beat me to it lol.

I noticed in the make file there's a '/usr/include/libcamera-apps' gate. I do have this directory.

jcfain, 2025-05-17:
Ok I seen you updated the make file thanks for that! 
I had to: 
`sudo apt install libboost-dev` 
and
`sudo apt libboost-program-options-dev`
to get it to build. Not sure if libboost-dev was needed.
I have the v4l2libcamera bin now on to how to use it...